RobotLocomotion/drake-ros #9: Dockerize drake-ros

IanTheEngineer, 2021-08-16:
This will aid in the creation of a CI pipeline. Create a dockerfile to the following requirements:

- [ ] base on prebuilt `ros:${ROS_DISTRO}-ros-base` docker container (`rolling` to start)
- [ ] install Drake and Drake's dependencies
- [ ] copy drake-ros source code into the container
- [ ] install source code dependencies
- [ ] build source code
- [ ] run unit tests, report results

EricCousineau-TRI, 2023-03-09:
Howdy Aditya!
I have no plan in the near term to execute it, so I would love to review your work if you have the time to try it out!

Can I ask your desired workflow? Do you want to have the container on DockerHub so you can easily fetch / build on top of it?
Or do you just want to have a Dockerfile / script that you can run locally to build from scratch?

For the record, I've *loved* using Apptainer for my own dev workflow of testing things out. Here's a rough cut of the workflow (that I stole from @sloretz):
https://github.com/EricCousineau-TRI/repro/tree/master/shell/apptainer_stuff

Kzernobog, 2023-03-10:
I have mostly used Drake's python bindings on my local machine(mostly to setup proof of concepts and simple simulations). I am thinking of combining this and the [drake_ros](https://github.com/RobotLocomotion/drake-ros) package as part of my research setup with actual robots, mostly to try and push drake's realtime use case on actual robots. We have predominantly used docker as part of our lab's containerisation drive. I did use Apptainer (back when it was called singularity) briefly. Could you help me understand why you prefer Apptainer over docker?

Yes the idea is to have a standalone drake and drake_ros container that could be extended with further research code.

EricCousineau-TRI, 2023-03-10:
Gotcha, that all sounds great!

> [...] mostly to try and push drake's realtime use case on actual robots [...]

Awesome!

> We have predominantly used docker as part of our lab's containerisation drive.

Aye, I think this and adoption from wider community.

I think if your lab is already familiar and used to Docker, then I'd be happy to review that if that's what you'd like. And of course, I'm sure other people would be more interested at present in Docker than Apptainer, esp. given that:
- That's actually what Drake publishes to anywho, as you might've seen: https://drake.mit.edu/docker.html#stable-releases
- Apptainer can fetch Docker containers ;)

> Could you help me understand why you prefer Apptainer over docker?

Apologies if I mislead / wasn't clear.
I'm happy to review Docker-based workflow code, esp. for provisioning. I'm just an avid (rabid?) Apptainer proselyte ;)

Rationale below in collapsable section - feel free to ignore if it's noise to you for now!

<details>
<summary>Apptainer Rabies</summary>
First party docs: https://apptainer.org/docs/user/1.1/introduction.html#why-use-apptainer
Some of my own terse notes here: https://github.com/EricCousineau-TRI/repro/tree/master/shell/apptainer_stuff

My motivation may be due to my ignorance of Docker, but having futzed with it over the past few years, I've found it easier to "futz with Apptainer" than "futz with Docker".

Specifically:
- It doesn't aggressively containerize by default. I can `apt install` crazy things in the container, but still run things against my home directory. NVidia works in a relatively straightforward way.
- It's not a server / daemon (at least by default). When trying to reattach to a running container, I get massively confused if I accidentally pass the wrong `docker run` flags, etc. Things like `docker-compose` can be replaced by simpler bash / Python glue scripts.
- I can run `apptainer` as a *user* without any crazy `sudo` masquerading. (It may be easier to use this on computing where it may be difficult to get `sudo` access directly)

The maintainers are also relatively open to responding to questions, at least as I've seen it, e.g.
https://github.com/apptainer/apptainer/issues/945
</details>

Kzernobog, 2023-03-14:
Sounds great Eric! Thanks for your input.

adeeb10abbas, 2023-08-02:
@EricCousineau-TRI can we close this?

EricCousineau-TRI, 2023-08-02:
Yup, closed by #272!

closed 2023-08-02 by EricCousineau-TRI