RobotLocomotion/drake-ros #43: Improve `bazel_ros2_rules` error handling

hidmic, 2022-02-21:
Failure modes for `bazel_ros2_rules` are quite opaque, if at all traceable -- specially when it comes to errors during CMake configuration scraping. Overall UX needs improvement.

hidmic, 2022-02-21:
FYI @EricCousineau-TRI. For instance, if some package upstream isn't exporting its dependencies properly, we're in for a fun archaeological trip down here.

EricCousineau-TRI, 2022-02-21:
Oh nuts! Do you have example of how opaque errors currently are? Perhaps via `gist.github.com`?

hidmic, 2022-02-21:
See [here](https://gist.github.com/hidmic/f91e586a26ba07289b2354317a90233f). We basically have to cap noisy tracebacks, collect CMake server output to bundle it with any exception thrown, and repackage both in the outermost `try ... except` block for human consumption.