RobotLocomotion/drake-ros #51: Should separate Skylark repository rule computation from workspace-analysis rule computation

EricCousineau-TRI, 2022-03-26:
Per comment and benchmarking in Anzu PR 8525:

> **Warning**: This has some high-overhead repository rule generation, meaning any
> BUILD files that depend on `load('@ros2//etc', ...)` will need to ensure the
> repository rule fully executes before those dependencies can be analyzed.
> Because of this, it may block other rules from effectively building.
> 
> For example, `bazel test --config=lint //...` will cause the repository rule to
> be fetched (and executed if needed).
> 
> To benchmark anew, run:
> ```sh
> cd anzu
> bazel clean --expunge --async
> # First, fetch stuff for simple Drake target to handle most repositories.
> bazel fetch @drake//:module_py
> # Now, record how long it takes for a ROS 2 target.
> time bazel fetch @ros2//:rosidl
> ```
> 
> On 2022-03-10, harddrives listed for `~/.cache/bazel`:
> 
> - Eric's Puget (nproc=64, 4TB SSD), took 60s
> - Eric's P15 Laptop (nproc=12, 1TB NVMe), took 40s

Ideally, if not actually building any ROS 2 dependencies, we should attempt to minimize how much time is required to at least allow the loading of skylar rules from `@ros2`.

To do so, we need to somehow separate the ROS 2 Skylar from the actual ROS 2 dependencies.

For this, I would propose:

- Separating Starlark wrapping code to be emitted into `@ros2_rules` (or `@ros2_bazel_rules`
- Keep the main code in `@ros2`

@sloretz Do you think this is something we can queue up for the next month or so?

sloretz, 2022-04-06:
> For this, I would propose:
> 
 > *   Separating Starlark wrapping code to be emitted into @ros2_rules (or @ros2_bazel_rules)

I'm not sure what these means, and I think I'm missing a concept. By starlark wrapping code, do you mean the code that invokes cmake-server and writes out a `BUILD.bazel` file? If code depends on `@ros2//:rclcpp_cc`, how does bazel know that target is supposed to exist without the `BUILD.bazel` existing yet?

 > *   Keep the main code in @ros2

Do you mean, commit the generated `BUILD.bazel` in the repo so it's not regenerated every time? Maybe a lint test could make sure the committed `BUILD.bazel` doesn't fall out of sync with the tarball?

EricCousineau-TRI, 2022-07-29:
>  If code depends on [...]

Er, yeah, upon rereading my statement above, doesn't make sense as written, sorry about that.

Perhaps what I should've said is, we should make it easy to swap between generating Starlark at build time or using pre-generated artifacts.

> Do you mean, commit the generated `BUILD.bazel ` [...]

Yes, or at least providing the option to do so. And per above,

jwnimmer-tri, 2023-03-09:
I'd advise against "swapping".  That's complexity trying to implement things that Bazel should be doing for us.

To me, instead the goal should just be to make it fast in the first place:
- Remove spurious dependencies.  There's no reason the `bazel_ros2_rules` skylark needs to load the whole ros2 archive before it can be analyzed.  The `ros_cc_binary` should come from `load("@rules_ros2//:ros_cc.bzl", "ros_cc_binary") not `@ros2`.
- Split downloading from file-generation (i.e., use a different repository _name_ for the archive itself vs the generated rules).  We want bazel to cache the extracted data for us, even when we've altered the repository rule code that post-processes it into something we can consume.
- Parallelize the codegen step using `multiprocessing`.