RobotLocomotion/drake-ros #53: Should prevent ROS_DISTRO from polluting build

EricCousineau-TRI, 2022-04-05:
As encountered in Anzu build, if a ROS 1 workspace / setup is sourced, then the build may fail, e.g.:
```sh
INFO: Repository ros2 instantiated at:
  {source}/WORKSPACE:56:25: in <toplevel>
  {source}/tools/workspace/default.bzl:282:24: in add_default_repositories
  {source}/tools/workspace/ros2/repository.bzl:53:26: in ros2_repository
Repository rule _ros2_repository_rule defined at:
  {source}/tools/workspace/ros2/repository.bzl:42:40: in <toplevel>
ERROR: An error occurred during the fetch of repository 'ros2':
   Traceback (most recent call last):
    File "{source}/tools/workspace/ros2/repository.bzl", line 27, column 25, in _ros2_repository_impl
        base_ros2_repository(
    File "{workspace}/external/bazel_ros2_rules/ros2/defs.bzl", line 87, column 29, in base_ros2_repository
        result = execute_or_fail(repo_ctx, cmd, quiet=True)
    File "{workspace}/external/bazel_ros2_rules/tools/execute.bzl", line 16, column 13, in execute_or_fail
        fail("Failed to setup @{} repository: {}".format(
Error in fail: Failed to setup @ros2 repository: './run.bash {workspace}/external/bazel_ros2_rules/ros2/generate_distro_file.py -s {workspace}/external/ros2/ros2-linux:ros2-linux -d distro_metadata.json ros2' exited with 1
--- captured stdout ---
ROS_DISTRO was set to 'noetic' before. Please make sure that the environment does not mix paths from different distributions.

--- captured stderr ---
Traceback (most recent call last):
  File "{workspace}/external/bazel_ros2_rules/ros2/generate_distro_file.py", line 105, in <module>
    main()
  File "{workspace}/external/bazel_ros2_rules/ros2/generate_distro_file.py", line 98, in main
    args = parse_arguments()
  File "{workspace}/external/bazel_ros2_rules/ros2/generate_distro_file.py", line 64, in parse_arguments
    args.distro = json.load(args.distro_file)
  File "/usr/lib/python3.8/json/__init__.py", line 293, in load
    return loads(fp.read(),
  File "/usr/lib/python3.8/json/__init__.py", line 357, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.8/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.8/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
ERROR: {source}/WORKSPACE:56:25: fetching _ros2_repository_rule rule //external:ros2: Traceback (most recent call last):
    File "{source}/tools/workspace/ros2/repository.bzl", line 27, column 25, in _ros2_repository_impl
        base_ros2_repository(
    File "{workspace}/external/bazel_ros2_rules/ros2/defs.bzl", line 87, column 29, in base_ros2_repository
        result = execute_or_fail(repo_ctx, cmd, quiet=True)
    File "{workspace}/external/bazel_ros2_rules/tools/execute.bzl", line 16, column 13, in execute_or_fail
        fail("Failed to setup @{} repository: {}".format(
```

\cc @IanTheEngineer

IanTheEngineer, 2022-04-05:
There must be some issue beyond the ROS1 environment being sourced. I completed builds with ROS1 sourced and unsourced without encountering this issue. I was only able to reproduce this when using a bazel environment built from a branch  that contained the `bazel_ros2_rules`, then switching to and attempting to build a branch that did not contain the `bazel_ros2_rules` external package (without a `bazel clean` first). `bazel clean` before building the branch after the switch resulted in a successful build. We first need to root cause this issue.

IanTheEngineer, 2022-04-05:
As discussed out-of-band with @EricCousineau-TRI, this is a lower priority issue.

EricCousineau-TRI, 2022-04-05:
> I was only able to reproduce this when using a bazel environment built from a branch that [...]

Bazel attempts to be hermetic, but it's not always perfect. IMO, this isn't entirely surprising, and I think we can either place fixes or mitigations to address it.

> We first need to root cause this issue.

I think the stacktrace is sufficient root cause, at least for this acute failure?

EricCousineau-TRI, 2022-04-05:
Ah, you mean "root cause why `bazel clean` didn't cause failure".

I think in this case, you forgot to source ROS 1 in the environment.

I am able to reproduce this error in Anzu by doing `bazel clean --expunge --async`, then doing `bazel fetch @ros2//:rosidl`.

EricCousineau-TRI, 2022-04-05:
I can reproduce this error in this repository under similar circumstances, using 685301d

hidmic, 2022-04-05:
Well, I keep getting notifications (should probably unwatch) so I'll chime in and hopefully save some of @sloretz's time :smile: 

> ROS_DISTRO was set to 'noetic' before. Please make sure that the environment does not mix paths from different distributions.

That message is likely coming from https://github.com/RobotLocomotion/drake-ros/blob/685301d189caef09b3599ddc8718d3043b6a0bbe/bazel_ros2_rules/ros2/resources/templates/run.bash.in#L14 The same message finds its way into the metadata file in [the previous step]((https://github.com/RobotLocomotion/drake-ros/blob/685301d189caef09b3599ddc8718d3043b6a0bbe/bazel_ros2_rules/ros2/defs.bzl#L78)), which causes all subsequent parsing to break.

---

One solution would be to nuke the environment in [`run.bash.in`](https://github.com/RobotLocomotion/drake-ros/blob/685301d189caef09b3599ddc8718d3043b6a0bbe/bazel_ros2_rules/ros2/resources/templates/run.bash.in) before all sourcing. Alternatively, you could use the `-o`/`--output` flag in all `ros2bzl`-powered scripts instead of [piping their stdout to a file](https://github.com/RobotLocomotion/drake-ros/blob/685301d189caef09b3599ddc8718d3043b6a0bbe/bazel_ros2_rules/ros2/defs.bzl#L78) through Bazel (at the time I believed it was connected to with spurious repo invalidation, but maybe I was wrong).

EricCousineau-TRI, 2022-04-05:
Ah, I think I get - this is a warning that is sent to stdout (ewww). Using `-o` allows us to actually capture what we want, rather than letting this warning scribble into our stdout input.

Do we know what code is responsible for this? Can we file an issue / PR to make that go into stderr instead?
(or is it an issue w/ our code?)

sloretz, 2022-04-05:
> Do we know what code is responsible for this? Can we file an issue / PR to make that go into stderr instead?
(or is it an issue w/ our code?)

I think it's from here:

https://github.com/ros/ros_environment/blob/64678bac5fd733146ba6d07a38d1db733398898e/env-hooks/0.ros_distro_check.sh.in#L3-L5

sloretz, 2022-04-05:
> Can we file an issue / PR to make that go into stderr instead?

Opened PR: https://github.com/ros/ros_environment/pull/30

sloretz, 2022-04-06:
I think this is resolved by #54; closing.

PR #54 (2022-04-05): Use --output so ROS_DISTRO being set doesn't affect build
Fixes #53 thats to @hidmic's comment https://github.com/RobotLocomotion/drake-ros/issues/53#issuecomment-1088740125

@IanTheEngineer & @EricCousineau-TRI

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake-ros/54)
closed 2022-04-06 by sloretz