RobotLocomotion/drake-ros #63: bazel_ros2_rules generates invalid BUILD.bazel when there's an underlay

sloretz, 2022-04-12:
Currently when an existing local ROS 2 workspace is sourced, `bazel_ros2_rules` will generate a BUILD.bazel file with absolute paths in places where Bazel labels are expected, thus making the whole build file invalid.

```console
ros2_example_bazel_installed$ bazel build //...
Starting local Bazel server and connecting to it...
ERROR: Traceback (most recent call last):
	File "/home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/BUILD.bazel", line 483, column 16, in <toplevel>
		share_filegroup(
	File "/home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/tools/common.bzl", line 8, column 45, in share_filegroup
		srcs = [path for path in native.glob(
Error in glob: pattern cannot be absolute
ERROR: /home/sloretz/ws/drake-ros/drake-ros/ros2_example_bazel_installed/ros2_example_common/BUILD.bazel:6:24: no such target '@ros2//:action_msgs_defs': target 'action_msgs_defs' not declared in package '' defined by /home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/BUILD.bazel and referenced by '//ros2_example_common:_ros2_example_common_msgs__rosidl_typesupport_cpp_gen'
ERROR: /home/sloretz/ws/drake-ros/drake-ros/ros2_example_bazel_installed/ros2_example_common/BUILD.bazel:6:24: no such target '@ros2//:builtin_interfaces_defs': target 'builtin_interfaces_defs' not declared in package '' defined by /home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/BUILD.bazel and referenced by '//ros2_example_common:_ros2_example_common_msgs__rosidl_typesupport_cpp_gen'
ERROR: /home/sloretz/ws/drake-ros/drake-ros/ros2_example_bazel_installed/ros2_example_common/BUILD.bazel:6:24: every rule of type rosidl_generate_genrule implicitly depends upon the target '@ros2//:rosidl', but this target could not be found because of: no such target '@ros2//:rosidl': target 'rosidl' not declared in package '' defined by /home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/BUILD.bazel
ERROR: Analysis of target '//ros2_example_common:_ros2_example_common_msgs__rosidl_typesupport_cpp_gen' failed; build aborted: Analysis failed
```

<details><summary>Examples in BUILD.bazel</summary>

```Python
share_filegroup(
    name = "rti_connext_dds_cmake_module_share",
    share_directories = ["/home/sloretz/ws/drake-ros/ros2/install/share/rti_connext_dds_cmake_module", "/home/sloretz/ws/drake-ros/ros2/install/share/ament_index"],
)
```

```Python
cc_library(
    name = "rmw_fastrtps_dynamic_cpp_cc",
    srcs = ["/home/sloretz/ws/drake-ros/ros2/install/lib/librmw_fastrtps_dynamic_cpp.so", "/home/sloretz/ws/drake-ros/ros2/install/lib/libfastrtps.so.2.6"],
    hdrs = glob(["{}/**/*.h*".format(x) for x in ["archive/rmw_fastrtps_dynamic_cpp"]]),
    includes = ["archive"],
    copts = ["-isystem /home/sloretz/ws/drake-ros/ros2/install/include/rmw_fastrtps_dynamic_cpp"],
    defines = [],
    linkopts = ["-Wl,-rpath,/home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/archive/rosidl_typesupport_cpp/lib:/home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/archive/rosidl_typesupport_c/lib:/home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/archive/rcpputils/lib:/home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/archive/rcutils/lib:/home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/archive/rmw/lib:/home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/archive/rmw_dds_common/lib:/home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/archive/rmw_fastrtps_shared_cpp/lib:/home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/archive/rosidl_runtime_c/lib:/home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/archive/rosidl_typesupport_fastrtps_c/lib:/home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/archive/rosidl_typesupport_fastrtps_cpp/lib:/home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/archive/rosidl_typesupport_introspection_c/lib:/home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/archive/rosidl_typesupport_introspection_cpp/lib:/home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/archive/fastcdr/lib:/home/sloretz/ws/drake-ros/ros2/install/lib:/home/sloretz/.cache/bazel/_bazel_sloretz/effac4167a518523c467e34b5d8c3dd7/external/ros2/archive/fastrtps/lib"],
    data = [":rmw_fastrtps_dynamic_cpp_share", "/home/sloretz/ws/drake-ros/ros2/install/lib/librmw_fastrtps_dynamic_cpp.so", "/home/sloretz/ws/drake-ros/ros2/install/lib/libfastrtps.so.2.6"],
    deps = [":rmw_dds_common_cc", ":rcutils_cc", ":rmw_fastrtps_shared_cpp_cc", ":rcpputils_cc", ":rosidl_runtime_c_cc", ":rosidl_typesupport_fastrtps_cpp_cc", ":rosidl_typesupport_introspection_c_cc", ":rosidl_typesupport_introspection_cpp_cc", ":rosidl_typesupport_fastrtps_c_cc", ":fastcdr_cc", ":fastrtps_cc", ":rmw_cc", ":fastrtps_cmake_module_cc"],
)
```

```Python

sloretz, 2022-04-12:
Possible fixes
* Invoke [ros2_local_repository](https://github.com/RobotLocomotion/drake-ros/blob/7ac6b2fd8e22f293313a608545c0ceda75e554b4/bazel_ros2_rules/ros2/defs.bzl#L186) on the underlay workspace, and use those rules?
* Maybe we should filter out any underlay workspaces from environment variables?
* emit an error message if the ROS 2 workspace is already sourced?

EricCousineau-TRI, 2022-04-12:
> Maybe we should filter out any underlay workspaces from environment variables?

I think this makes the most sense.

Ideally, our Bazel rules should be as explicit as possible in *source code*, and not change behavior based on envrionment variables, esp. if the goal is to explicitly list the chain of workspaces / overlays.
If a user wants to use a workspace, they should explicitly list in their `WORKSPACE`. If they really want to be at the behest of their environment variables, they can use tricks like Anzu's `os.bzl` to load in env vars.

What do you think?