RobotLocomotion/drake-ros #77: Should add example usage of `rmw_isolation` in `ros2_example_apps/test`

EricCousineau-TRI, 2022-05-04:
Motivated by @sloretz's question in #75 - https://github.com/robotlocomotion/drake-ros/pull/75#pullrequestreview-961180890

At present, Miche's awesome isolation work is presently only tested / shown in `bazel_ros2_rules`. For simplicity, we should show simple examples of isolation in a more app-oriented usage (and less about unit/stress-testing it).

C++
https://github.com/RobotLocomotion/drake-ros/blob/e34b93b396e8af7ae938f9ce0f3ac775488dc0e5/bazel_ros2_rules/ros2/resources/rmw_isolation/test/isolated_talker.cc#L32-L35
https://github.com/RobotLocomotion/drake-ros/blob/e34b93b396e8af7ae938f9ce0f3ac775488dc0e5/bazel_ros2_rules/ros2/resources/rmw_isolation/package.BUILD.bazel#L32-L47

Python
https://github.com/RobotLocomotion/drake-ros/blob/e34b93b396e8af7ae938f9ce0f3ac775488dc0e5/bazel_ros2_rules/ros2/resources/rmw_isolation/test/isolated_talker.py#L23-L26
https://github.com/RobotLocomotion/drake-ros/blob/e34b93b396e8af7ae938f9ce0f3ac775488dc0e5/bazel_ros2_rules/ros2/resources/rmw_isolation/package.BUILD.bazel#L62-L73

This reflects our LCM-style testing philosophy - we explicitly let tests denote how they wish to isolate, but provide utilities to facilitate it.

EricCousineau-TRI, 2022-05-04:
Will see if I can do this in ~10min - if not, @sloretz, can I pass this to you?

sloretz, 2022-05-04:
> Will see if I can do this in ~10min - if not, @sloretz, can I pass this to you?

I can give it a shot. I could use the practice with it.

EricCousineau-TRI, 2022-05-04:
Sounds great - thanks!!!

EricCousineau-TRI, 2022-05-04:
For both C++ and Python, I'd suggest just keeping it single-node, and perhaps just do a single-process talker / listener.

No need to stress test here, as I believe Miche already handled that in `@ros2`

EricCousineau-TRI, 2023-02-16:
Closed as duplicate of #115

closed 2023-02-16 by EricCousineau-TRI