RobotLocomotion/drake-ros #78: Should enable drake-from-source build on CI (for Bazel build testing for C++ and Python)

EricCousineau-TRI, 2022-05-04:
From Drake Slack conversations:
- In CMake, we can do C++ and Python bindings from a binary (or source)
- In Bazel-from-source (`drake_bazel_external`), we can do C++ and Python bindings from source
- In Bazel-from-binary (`drake_bazel_installed`), we can do *very unsupported* C++, and Python *code* (but *not* bindings). Per Slack, there is no plan to prioritize this build mode.

Given this, we should enable source build on CI, ideally using Drake release pins + caching to minimize CI overhead + cost.

Per Jeremy's suggestion, we should try to fit w/in GitHub actions budget.
Failing that, we spend $$ (fastest) to provide sufficient CI resources to do this, as Drake budget is already fully allocated. Alternative is to help make Drake CI more efficient; however, that has a bit more overhead.

FYI @IanTheEngineer

EricCousineau-TRI, 2022-05-04:
@sloretz Are you OK w/ me assigning this to you? Do you need help / pointers on configuring Bazel cache?

EricCousineau-TRI, 2022-05-09:
Pointer to Jeremy's advice on Drake Slack: https://drakedevelopers.slack.com/archives/C0KDGF4V9/p1651623384081889

EricCousineau-TRI, 2022-05-10:
@sloretz One thing I'm not sure about - we may want to schedule a weekly build to populate cache... have you thought about that / seen an examples from your investigations?

sloretz, 2023-01-06:
I think this ticket might be complete. The `drake_ros_*` bazel tests [do build Drake from source using a pinned version](https://github.com/RobotLocomotion/drake-ros/blob/a4d976d6ef093031b4a08e31446527c814882501/drake_ros_core/WORKSPACE#L43-L45), and the bazel CI [does use a persistent cache](https://github.com/RobotLocomotion/drake-ros/blob/a4d976d6ef093031b4a08e31446527c814882501/.github/workflows/bazelized.yml#L16-L19).  

Ok to close this one?

EricCousineau-TRI, 2023-01-10:
Yup! Will close.

Note for future readers: I forgot what I meant by "very unsupported" in above post, so have struck it out.

closed 2023-01-10 by EricCousineau-TRI