RobotLocomotion/drake-ros #90: Migrate bazelized github action away from ros-rolling-*-focal docker image

sloretz, 2022-06-09:
The bazelized workflow starts from a ROS Rolling focal image, but as Rolling rolled onto jammy I don't think it will get any more updates.

https://github.com/RobotLocomotion/drake-ros/blob/389c372f06179a04d6778d73bfe9e8d2187a627b/.github/workflows/bazelized.yml#L13

The test steps don't actually use the ROS Rolling workspace that comes with it. They download ROS from the `drake-ros-underlay` archive. It seems like the image is being used for ROS tools (like rosdep). I think the job could use a plain Docker image (or the Drake focal image) with [setup-ros](https://github.com/ros-tooling/setup-ros) to get the ROS tooling.

EricCousineau-TRI, 2023-01-09:
Is this possibly why we have CI failures? e.g.
https://github.com/RobotLocomotion/drake-ros/actions/runs/3802593789/jobs/6468228156
https://github.com/RobotLocomotion/drake-ros/actions/runs/3832224473/jobs/6522290695

sloretz, 2023-01-09:
Both of those should be fixed by #204, though maybe the docker image they start from is to blame.

EricCousineau-TRI, 2023-01-10:
Gotcha! Yeah, could we change that image to leverage Jammy?