RobotLocomotion/drake-ros #95: Should remove as many hard-coded paths from generated shell files

EricCousineau-TRI, 2022-07-28:
At present, I see things like
```
for workspace in ~/.cache/bazel/_bazel_{user}/{hash{/external/ros2/ros2-linux; do ...
```

I don't believe we should be using absolute paths like this.
I suggest we use `${BASH_SOURCE}` or something like `$(readlink -f ${BASH_SOURCE})` instead.

cottsay, 2022-08-26:
I believe you're referring to this?
https://github.com/RobotLocomotion/drake-ros/blob/f5e455a51dc6660c90cee0db2fb1a20fa345953b/bazel_ros2_rules/ros2/resources/templates/run.bash.in#L11

When I use a ROS installation in `/opt/ros/humble` instead of letting Bazel download and extract the tarball, `@WORKSPACE@` gets resolved to that. So it appears that `~/.cache/bazel/_bazel_{user}/{hash}/external/ros2/ros2-linux` just happens to be where the tarball gets extracted to. I'm not sure we can avoid the absolute path here.

EricCousineau-TRI, 2022-08-26:
I believe we can use `os.path.relpath` in Python in conjunction with `$(dirname ${BASH_SOURCE})` to permit relative paths where appropriate (when extracted locally vs. scanned from system).