RobotLocomotion/drake-ros #104: dds, multicast: Documentation for multicast needs better troubleshooting steps / cross references

EricCousineau-TRI, 2022-08-02:
In addition to https://github.com/ros2/ros2_documentation/issues/2615, the documentation for debugging multicast may need to be expanded.

From investigations on separate codebase, here are a suite of relevant commands for debugging on Ubuntu, hositing from Anzu issue 9239

---

Experiment setup, both machines Ubuntu 20.04:
- Desktop, connected to internet. Reserve one Ethernet NIC for laptop. Can configure for shared.
- Laptop, disconnected from everything. Plug NIC into Desktop NIC. Configure for Internet (Automatic DHCP), or Manual. Disconnect WiFi so there's only one possible multicast route.

Can use whatever UDP multicast, either `ros2 multicast send/receive`, or LCM.
Simple LCM script: https://github.com/RobotLocomotion/drake/commit/fd50a9e61

Debugging commands
```
# see connections - raw to nicer to see
ip addr
ifconfig
nmcli

# see routes
ip route
# see specific route
ip route get <multicast-ip>

# add multicast route
sudo ip route add 224.0.0.0/4 dev <dev>
# remove multicast route
sudo ip route del 224.0.0.0/4 dev <dev>

# if using nm-connection-editor, add for 224.0.0.0, netmask 240.0.00 (/4)

# er... something
sudo iptables -L

# see existing connections - e.g. check if local address is `0.0.0.0` or specific bound address
netstat -anu

# look at connetions via wireshark, w/ display filter and autostart
sudo wireshark -i <dev> -Y 'ip.proto == "udp"' -k
```

Some connection notes:

### Sender: Desktop, Receiver: Laptop

If both connected to LAN, should work - same as below.

If not:
- No routes - no dice.
- With route on Desktop, not (explicitly) on Laptop: Will route through "default" routing.
- With rout eon Desktop, explicit: Still good.

### Sender: Laptop, Receiver: Desktop

If Laptop connected to LAN, ensure both machines have route to multicast (via `ip route get 224.0.0.0`). Then it should work, as "default" routing should route multicast.

If not, then:
- No routes - no dice.
- With route on Laptop, not Destkop: Can see traffic on Wireshark, but LCM won't receive b/c LCM receive socket will bind to the concrete address (and then get confused?. (Per Sammy, better than binding to `0.0.0.0:port` and then filtering all traffic.)
- With route on both, works.

### Additional Notes

Cannot explicitly route through multiple interfaces (w/o bridging), I believe.
Though it seems like kernel allows "default" to be shadowed - e.g. switching multicast from Internet (office LAN) to a more specific device.

EricCousineau-TRI, 2022-08-02:
\cc @sloretz @IanTheEngineer

EricCousineau-TRI, 2022-08-05:
FYI: See semi-automated physical network connection testing in https://github.com/EricCousineau-TRI/franka_panda_udp_timing

automates Wireshark via `tshark`, dispatches commands through SSH, etc.

EricCousineau-TRI, 2022-08-08:
See also #103, https://github.com/ros2/ros2_documentation/issues/2615, https://github.com/RobotLocomotion/drake-ros/issues/99

IanTheEngineer, 2022-08-08:
Tooling from Scott: https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Getting-Started-With-Ros2doctor.html