RobotLocomotion/drake-ros #99: corss-talk check: Sometimes get error message about "lo" not being multi-cast capable

EricCousineau-TRI, 2022-08-02:
@shane-loretz-ctr On certain machines, if I run `./run //tools:ros2 node list`, I sometimes get the following warning:
```
ros2: selected interface "lo" is not multicast-capable: disabling multicast
```
However, this only happens once per machine.
It seems to have happened on only 2 of 3 machines (I think...), which is disturbing, but I want to check.

How do I clear the cache for this warning / make it always show up across invocations?

EricCousineau-TRI, 2022-08-02:
(FTR, I assume this is part of `RMW_LOCALHOST_ONLY` working, but also want to confirm)

sloretz, 2022-08-02:
> However, this only happens once per machine.
> It seems to have happened on only 2 of 3 machines (I think...), which is disturbing, but I want to check.

The warning comes from [this line in CycloneDDS](https://github.com/eclipse-cyclonedds/cyclonedds/blob/0e2cd3e303be2171dd0e4fc685cc5031f70b0f52/src/core/ddsi/src/q_init.c#L1426).  On my machine I can make the warning disappear with `sudo ifconfig lo multicast`, and reappear with `  sudo ifconfig lo -multicast`. Communication between two nodes with `ROS_LOCALHOST_ONLY=1` seems to work fine whether multicast is enabled on the loopback device or not.

Maybe some machines don't see the warning because they were previously [ configured to use LCM without a network interface](https://lcm-proj.github.io/multicast_setup.html)?

I'd recommend enabling multicast on the loopback device to silence the warning, but if that isn't sufficient I can look deeper at cyclonedds's logging mechanism.

EricCousineau-TRI, 2022-08-02:
Gotcha; if things work fine on a single machine w/ or w/o multicast loopback, then great!
That being said, we should be careful about how we establish multicast. I am tentative to enable by default w/o understanding implications - e.g. how it may impact doing real multicast in the future in an intentional fashion.

But I am still *very* concerned on the warning appearing once the same machine.
Why does this happen?
Is it non-determinism? (like it's just selecting a device at random?)
Is it a persistent log-once kind of thing? etc

sloretz, 2022-08-02:
> But I am still very concerned on the warning appearing once the same machine.

I haven't been able to reproduce this. On my machine the warning appears once at the start of every ROS 2 process when multicast is disabled on the loopback device. I tried restarting processes multiple times, and launching other processes but the behavior has been the same every time. When I enable multicast the warning never appears.

I'm using the commands `ROS_LOCALHOST_ONLY=1 bazel run //ros2_example_apps:oracle_cc`, `ROS_LOCALHOST_ONLY=1 ros2 topic echo chatter std_msgs/msg/String`, and `ROS_LOCALHOST_ONLY=1 ros2 topic pub chater std_msgs/msg/String "data: 'foobar'"` for my testing. Do you see inconsistent warning behavior with those?

IanTheEngineer, 2022-08-02:
| But I am still very concerned on the warning appearing once the same machine.

Can you elaborate? Does this happen once per machine or multiple times on the same machine?

EricCousineau-TRI, 2022-08-02:
I only saw the message once per machine - that is, if I run `./run //tools:ros2 node list` 10 times, I only saw it the very first time.
This happened on two machines (my laptop and office Puget), but not on a third (robot Puget).

I am unable to produce, but I know that I received the warning b/c I copied and pasted it here.

I will try: `env ROS_LOCALHOST_ONLY=1 bazel-bin/external/ros2/ros2 node list` using `drake-ros` with #101

EricCousineau-TRI, 2022-08-02:
I can intermittently reproduce. This is using 8c20da4, above command

Robot Puget:
```
bash-5.0$ env ROS_LOCALHOST_ONLY=1 bazel-bin/external/ros2/ros2 node list
bash-5.0$ env ROS_LOCALHOST_ONLY=1 bazel-bin/external/ros2/ros2 node list
bash-5.0$ env ROS_LOCALHOST_ONLY=1 bazel-bin/external/ros2/ros2 node list
bash-5.0$ env ROS_LOCALHOST_ONLY=1 bazel-bin/external/ros2/ros2 node list
bash-5.0$ env ROS_LOCALHOST_ONLY=1 bazel-bin/external/ros2/ros2 node list
bash-5.0$ env ROS_LOCALHOST_ONLY=1 bazel-bin/external/ros2/ros2 node list
bash-5.0$ env ROS_LOCALHOST_ONLY=1 bazel-bin/external/ros2/ros2 node list
```
```

EricCousineau-TRI, 2022-08-02:
All this being said, we can punt on this for the time being, and revisit if/when it becomes a problem.
I'm still disturbed about the spurious nature of it, but :shrug: for now.

EricCousineau-TRI, 2022-08-02:
In the FWIW category, I was able to recreate the warning if I rebooted machine.

IanTheEngineer, 2022-08-02:
I can confirm that I see the same results as you:

First call:
```
$ ./run //tools:ros2 node list
+ ./run //tools:ros2 node list
1659476179.279749 [0]       ros2: selected interface "lo" is not multicast-capable: disabling multicast
```
Every other time.
```
$ ./run //tools:ros2 node list
+ ./run //tools:ros2 node list
```
And setting the `ROS_LOCALHOST_ONLY` returns nothing:
```
$ env ROS_LOCALHOST_ONLY=1 bazel-bin/external/ros2/ros2 node list
```

EricCousineau-TRI, 2022-09-16:
As of late, possibly due to no mixing of `ros2cli` between `ROS_LOCALHOST_ONLY` modes, this became consistent.

Additionally, as indicated by Shane in https://github.com/eclipse-cyclonedds/cyclonedds/issues/1400, we can suppress this by using the nice startup script he mentioned in his overview comment.

Closing for now (I believe we have the "please warn bout ros2cli daemon settings mismatch" tracked in #103)

closed 2022-09-16 by EricCousineau-TRI