RobotLocomotion/drake-ros #107: bazel: Should admit meta testing of `bazel build` via `bazel test` itself

EricCousineau-TRI, 2022-08-03:
Per @jwnimmer-tri's suggestion in https://github.com/RobotLocomotion/drake-ros/pull/106#issuecomment-1204526002, we should be able to test the *build workflows* itself, e.g. ensuring our current shims do not corrupt Runfiles resolution (#105).

For now, we can just add specific CI testing, but we should eventually use Bazel for the niceties that it can provide.

\cc @IanTheEngineer @sloretz

jwnimmer-tri, 2022-08-03:
FYI Drake has one test case that does this already, see here:

https://github.com/RobotLocomotion/drake/blob/master/tools/install/bazel/test/drake_bazel_installed_test.py

It's a giant pain because the subprocess run of `bazel` does not (and cannot) use any caching, so it needs to download the `@rules_python` and `@remote_jdk11_tools` and all of that stuff every single time it runs, which tends to take a long time and sometimes has internet-flakes.

EricCousineau-TRI, 2022-08-03:
Ah, yeah, forgot about caching... Would be nice if Bazel offered an affordance for that.
Will icebox this for now, but at least there are TODO's 'n such.

jwnimmer-tri, 2022-08-03:
Yeah.  You can pre-populate the nested bazel's repository_cache via https://bazel.build/reference/command-line-reference#flag--distdir but then you need to get the pre-populated data _yourself_ somehow, but you can't hermetically access your _own_ repository cache.  So then you're in the land of genrules that fetch http files.  All doable, just a lot of cruft for just one test case.  (If you had a _lot_ of acceptance tests, you could probably automate all of the WORKSPACE and prefetch boilerplate, but not worth it for just one test.)

EricCousineau-TRI, 2022-08-03:
Aye, yeah, that makes sense. In this case, will mark this as closed for now; can reopen if need be.
Thanks!

closed 2022-08-03 by EricCousineau-TRI