RobotLocomotion/drake-ros #108: dload: Clean up items

EricCousineau-TRI, 2022-08-04:
- [ ] Minimize generated code. At present, for C++, the [template code is ~80 lines long](https://github.com/RobotLocomotion/drake-ros/blob/1e4da97bc8ad15a0a595e29c2ea390526dd8b3ed/bazel_ros2_rules/ros2/tools/dload_cc.bzl#L14-L94), with character escaping. For both C++ and Python, we should break this down and provide libraries that are then used via `ros_cc` and `ros_py`.
- [ ] Do not re-shim environment variables. At present, there are no checks that we're already shimming. We should ideally say something like `_BAZEL_ROS_RULES_SHIM`. If set, we should not inject *any* new environment variables (that should be controlled by entrypoint, similar to runfiles as shown in #106). If unset, then we set env vars, and set this sentinel. This should happen in both C++ and Python.

\cc @IanTheEngineer @sloretz

sloretz, 2022-08-09:
The second bullet point is in #110

sloretz, 2022-09-01:
The first bullet point is in #130

sloretz, 2022-09-29:
Closing as completed

PR #130 (2022-09-16): Add C++ and Python dload shim libaries to minimize generated code
Resolves #108

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake-ros/130)
closed 2022-09-29 by sloretz