RobotLocomotion/drake-ros #115: ros2_example_bazel_installed: Should have simple cc,py example of using RMW isolation

EricCousineau-TRI, 2022-08-10:
Per Anzu 9275, we should have a simple cc and py test that exercises RMW isolation.
e.g. `rmw_isolation_example_cc_test`, `rmw_isolation_example_py_test`

Both can simply publish a float, and do no other verification (for now).

They should then be referenced by:
https://github.com/RobotLocomotion/drake-ros/tree/eba927fd3ecf0ce2819200e0a8176e7d3f366e2a/bazel_ros2_rules/ros2#tools

\cc @IanTheEngineer

adityapande-1995, 2023-02-13:
I see a test here for isolated listener and talker in both C++ and python : https://github.com/RobotLocomotion/drake-ros/blob/main/bazel_ros2_rules/ros2/resources/rmw_isolation/test/isolated_listener.py. To we require something more to resolve this issue ? cc @sloretz

adityapande-1995, 2023-02-13:
Also a duplicate issue probably : https://github.com/RobotLocomotion/drake-ros/issues/77

sloretz, 2023-02-13:
> Do we require something more to resolve this issue ? cc @sloretz

I think I recall this issue was to have an example test in the `ros2_example_bazel_installed` folder.

That folder is a separate bazel workspace, which matches what our downstream users would use. The C++/Python files will probably look similar, so IIUC the value in putting tests there is in showing what bazel rules to `load()` and run in the `WORKSPACE` and `BUILD.bazel` files.

EricCousineau-TRI, 2023-02-16:
Yes, a very clear, user-based example of both C++ and Python of how to use isolation is my intent.

Having it in a more focused example workspace makes sense too! note that existing rmw_isolation tests are more for unittesting, less for exemplars.

closed 2023-04-28 by EricCousineau-TRI