RobotLocomotion/drake-ros #118: Workflow unclear to make `ros2 topic echo` work for custom-generated types

EricCousineau-TRI, 2022-08-16:
In Anzu, unable to use it for custom generated type; not immediately clear to me how to make this work.

FYI @sloretz @cottsay 
\cc @IanTheEngineer @calderpg-tri

EricCousineau-TRI, 2022-08-16:
It seems like the right workflow is to ensure the Python messages are on `PYTHONPATH` when running the `ros2` CLI:
3ffdb67
That at least makes `ros2 topic echo` work.

However, `ros2 interfaces list` does not work: 71c9e27
How do we update this to ensure that our messages etc. are discoverable?

sloretz, 2022-08-16:
> How do we update this to ensure that our messages etc. are discoverable?

The messages are made discoverable via the ament index. This is normally created via [this bit of CMake code](https://github.com/ros2/rosidl/blob/b8381d955a111cdf8a52a0f1891cc55de5f19db1/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake#L236-L252). There doesn't seem to be an equivalent in `bazel_ros2_rules`, but maybe [it could exist here](https://github.com/RobotLocomotion/drake-ros/blob/416cf8b81ad3e9c2d6855e2f43a5380328401ffa/bazel_ros2_rules/ros2/rosidl.bzl#L157) so that depending on `:custom_messages_defs` gave a target the ability to discover those messages. If so the key pieces are to generate ament_index files for custom messages, and to make the shim update the `AMENT_PREFIX_PATH` environment variables to point to those files when `:custom_messages_defs` is depended upon.

sloretz, 2022-08-29:
Fixed by #124

PR #124 (2022-08-26): Create ament index for custom messages and make dload shims aware of it
Fixes #118 
Contains #120

This creates an ament resource index describing the existence of custom messages created with `rosidl_interfaces_group()`. The index is made available as runfiles to any target that depends on `{group name}_ament_index`. The dload shim uses a bazel aspect to accumulate declared ament index prefixes, and prepends `AMENT_PREFIX_PATH` with them.

Note that the `ros2 interface list` test in `custom_message_echo_test` now passes because it depends on `:ros2_example_apps_msgs_ament_index`

- - -
This c [... cut]
closed 2022-08-29 by sloretz