RobotLocomotion/drake-ros #122: RViz2 has confusing behaviors - tf2 timing,

EricCousineau-TRI, 2022-08-19:
See repro in #121 
Specifically, this commit / README:
https://github.com/RobotLocomotion/drake-ros/blob/bcf2eeddd355dd3e73a9cb16ccd6b91905a18138/ros2_example_bazel_installed/repro/README.md

sub issues; I'd like help with confirming these issues, dispatching this to relevant repositories and the relevant fixes:
- [ ] Rviz2 is sensitive to only a few millseconds delay; it seems to have no option to control what lookup should be (timeout, use TimePointZero, etc.)
- [ ] There is millseconds delay between publishing and RViz receiving it - why?
- [ ] Rviz2 seems to not be showing all axes in `MarkerArray` - why?
- [ ] Why does `rclpy.spin_once(node)` freeze when a node has no subscribers? That seems awkward?

For any of these, we really should not encourage any one to try and publish markers in the future, nor have them try and publish `Time(0, 0)`.

It should be on the onus of the listener (Rviz) to handle this. My ideal is to have option to use `TimePointZero`, or whatever lookup means "give me the latest for this transform and its constituent lookups".

\cc @IanTheEngineer @sloretz @calderpg-tri @gbiggs

EricCousineau-TRI, 2022-08-29:
Per VC and [Drake Slack thread](https://drakedevelopers.slack.com/archives/C01B5B320QY/p1661802274172459), `rclpy.spin_once(node)` freezes because with `timeout_sec=None` means it will block until it has at least one unit of work (callback) available to execute (regardless of number of subscribers).

sloretz, 2022-08-29:
On item 3:

> Rviz2 seems to not be showing all axes in MarkerArray - why?

The fix is to give each marker a unique id: https://github.com/EricCousineau-TRI/drake-ros/pull/2/commits/8b0910f0130a2339f51587cd45f82f8736496012 . Markers in Rviz have a lifetime and can be manipulated. They're [uniquely identified via a namespace and id](https://github.com/ros2/common_interfaces/blob/57f156778da4cf999a0b24dc7732fc2b3bbb0b93/visualization_msgs/msg/Marker.msg#L27-L31). `Marker` messages [specify an action](https://github.com/ros2/common_interfaces/blob/57f156778da4cf999a0b24dc7732fc2b3bbb0b93/visualization_msgs/msg/Marker.msg#L34-L39), and [the default action is "add/modify"](http://design.ros2.org/articles/interface_definition.html#default-values). It will add a new marker with the given (namespace, id) if one doesn't already exist, otherwise it will modify the existing marker. Only the last axis is showing because each `Marker` message has the same (namespace, id), so one "marker" is added and then modified twice. This is probably not what someone wants when specifying multiple markers in a single `MarkerArray` message, so I'll look into adding a warning when this happens.

sloretz, 2022-08-29:
On item 3:

I opened a pull request to show an error status when a single `MarkerArray` message with a duplicate `ns` and `id` pair is received: https://github.com/ros2/rviz/pull/891

EricCousineau-TRI, 2022-08-30:
sweet, thanks! will check that box given your root cause and solution for warning

sloretz, 2022-08-30:
On Item 1:

> Rviz2 is sensitive to only a few millseconds delay; it seems to have no option to control what lookup should be (timeout, use TimePointZero, etc.)

Looking at #121's "via_tf", it looks like the issue is a bug in the Axes display. The ROS 1 to ROS 2 port changed the transform look up [from time point zero in ROS 1](https://github.com/ros-visualization/rviz/blob/6a234b4404c5ee71051f0b338ab8d5d2d931c27b/src/rviz/default_plugin/axes_display.cpp#L161) to [the current time in ROS 2](https://github.com/ros2/rviz/blob/4e976dc444e7e59ec9cdef30f1f04e0beafcafc8/rviz_default_plugins/src/rviz_default_plugins/displays/axes/axes_display.cpp#L105). I opened https://github.com/ros2/rviz/pull/892 fixing it, and I think it can be backported.

As to not having an option to control what the lookup should be, it looks like adding an option would require changes to [RViz's frame manager interface](https://github.com/ros2/rviz/blob/4e976dc444e7e59ec9cdef30f1f04e0beafcafc8/rviz_common/include/rviz_common/frame_manager_iface.hpp#L144-L182). If it's a priority I can work on it for ROS Rolling and Iron+, but that kind of change probably wouldn't be backportable to ROS Humble. I'm not sure what the size of that task would be.

I looked at how a few displays choose which transform time to lookup. I found only `Axes` had the bug where it tried to lookup the transform at the current time. Displays that are showing a ROS message with a Header use the transform at the time stamp in the header of the received message (things like point clouds, nav paths, grid cells, etc). Displays that don't have that (Grid, TF, etc) use the latest available transform.

EricCousineau-TRI, 2022-08-31:
> I opened https://github.com/ros2/rviz/pull/892 fixing it, and I think it can be backported.

Sweet, nice fix! Added some small comments there re: docs and dev advice.

> As to not having an option to control what the lookup should be, it looks like adding an option would require changes [...]

Ideally, it would "just" be a widget with configuration options common to allow RViz plugins consuming TF2 information.
If it's not too much, consider filing issue against rviz2? But yeah, `TimePointZero` prolly solves main use case, so we can let this possible improvement sit for a bit / get closed during cleaning if there's no further desire for it.

sloretz, 2022-09-01:
On Item 2:

> (2) There is millseconds delay between publishing and RViz receiving it - why?

I looked into the transit times using [ros2_tracing](https://github.com/ros2/ros2_tracing), a [simple python tf publisher](https://github.com/RobotLocomotion/drake-ros/files/9471878/tfdisplay_error.py.txt), and RViz using the Axes display. I'm pretty sure the delay is RViz's 30Hz refresh rate. Here's what I was able to get data for.

**No data**
1. Python process: Timestamp recorded via `node.get_clock().now()`
2. Python process: Python `tf_broadcast.sendTransform(message)`
3. Python process: `rclpy.publish(message)`

**Begin transit time measurement via [trace point in `rcl_publish()`](https://github.com/ros2/rcl/blob/191b13bc2eb8578e40799eb6b50c74b885e7066e/rcl/src/rcl/publisher.c#L248)**
4. Python process: `rcl_publish()`
5. Python process:  `rmw_publish()`
6. Python & RViz process: CycloneDDS delivers message from publisher process to RViz publisher
7. RViz process: `rmw_wait()`, `rcl_wait()` wakes
9. RViz process: executor takes message from middleware
11. RViz process: starts tf listener subscription callback

**End transit time measurement via [trace point in rclcpp callback dispatch](https://github.com/ros2/rclcpp/blob/df994e435df6cbdb19ab7ada839f403367fb769a/rclcpp/include/rclcpp/any_subscription_callback.hpp#L674)**
**Begin callback duration measurement via same tracepoint**
12. `TFListener` callback runs and puts new transform data into the tf buffer
**End callback duration measurement via [trace point at end of rclcpp callback dispatch](https://github.com/ros2/rclcpp/blob/df994e435df6cbdb19ab7ada839f403367fb769a/rclcpp/include/rclcpp/any_subscription_callback.hpp#L796)**

**No data**
13. RViz asks [`Axes` display to update](https://github.com/ros2/rviz/blob/dec5e8fafd57cfaa88eb9d871a4c16bfe9934d84/rviz_default_plugins/src/rviz_default_plugins/displays/axes/axes_display.cpp#L95) (this is subject to RViz's refresh rate)
14. Axes [gets the transform data from RViz's frame manager interface](https://github.com/ros2/rviz/blob/dec5e8fafd57cfaa88eb9d871a4c16bfe9934d84/rviz_default_plugins/src/rviz_default_plugins/displays/axes/axes_display.cpp#L104-L105).

With a sample size of 77 callbacks, the transit time from `rcl_publish` in the Python process to the beginning of the TransformListener callback in the RViz process was at most just under 0.5ms, but usually 0.1 to 0.2ms.

[image]

Then the TFListener callback runs. The first run took about 0.05 ms, but most were less than 0.01ms.

[image]

I think it's unlikely the extra time is coming from the unmeasured stuff on the Python side. Instead I think RViz's default refresh rate of 30Hz (33ms period) is the cause, and ros2/rviz#892 should make that no longer a problem.

sloretz, 2022-09-07:
@EricCousineau-TRI I think I've addressed all bullet points. If it looks good to you, may I close this ticket?

EricCousineau-TRI, 2022-09-08:
sweet, and yes! thanks a ton for the results!

50us - 200us seems high (from my naive perspective) for the small message you sent, but maybe that's related to the rclpy performance issues you mentioned earlier. good to know!

closed 2022-09-08 by EricCousineau-TRI