RobotLocomotion/drake-ros #125: cyclone dds: What does `/ParticipantIndex="none"` mean for usage in ROS2?

EricCousineau-TRI, 2022-08-26:
Wanted to check on rationale `ROS_DOMAIN_ID`, confirmed that yeah it's just direct link to DDS Domain ID, and it is used to partition port numbers.

Looked at docs for limit on number of participants:
https://docs.ros.org/en/humble/Concepts/About-Domain-ID.html#participant-constraints

Then also saw the Eclipse DDS stuff that @hidmic referenced to in `rmw_isolation`:
https://cyclonedds.io/docs/cyclonedds/0.9.1/config.html#controlling-port-numbers
https://github.com/eclipse-cyclonedds/cyclonedds/blob/0.9.1/src/core/ddsi/src/ddsi_portmapping.c#L39-L44
https://github.com/eclipse-cyclonedds/cyclonedds/blob/0.9.1/src/core/ddsi/include/dds/ddsi/ddsi_cfgelems.h#L1883-L1902
Looked at usage of `ParticipantIndex="none"` here too:
https://github.com/RobotLocomotion/drake-ros/blob/c7f9c362c5b4f2071235e61bc64f8ac1ad0c42f5/bazel_ros2_rules/ros2/resources/rmw_isolation/rmw_isolation.py#L153-L162

Questions:

1. I am a bit confused about when multicast or unicast is used. Are there ROS-specific docs about this? (may relate #103)
2. Do we need to set `ParticipantGain` at all if can use auto-resolve its port (and thus overcome limitations on participant IDs)? (from code linked above, it appears not to be)
3. Should this be mentioned to some extent in ROS documentation?

\cc @sloretz @cottsay

sloretz, 2022-09-09:
> I am a bit confused about when multicast or unicast is used. Are there ROS-specific docs about this?
> [...]
> Should this be mentioned to some extent in ROS documentation?

I don't know of any ROS specific documentation about when multicast or unicast is used, but it would be specific to a single RMW implementation. CycloneDDS has some documentation about when multicast is used here: https://cyclonedds.io/docs/cyclonedds/latest/config/cyclonedds_specifics.html#multicasting . The information is probably in the [DDSI-RTPS spec](https://www.omg.org/spec/DDSI-RTPS/2.5/PDF), but it would take a while to find it.

> Do we need to set `ParticipantGain` at all if can use auto-resolve its port (and thus overcome limitations on participant IDs)? (from code linked above, it appears not to be)

It looks like `ParticipantGain` [is a constant being set to the default value of 2](https://github.com/RobotLocomotion/drake-ros/blob/c7f9c362c5b4f2071235e61bc64f8ac1ad0c42f5/bazel_ros2_rules/ros2/resources/rmw_isolation/rmw_isolation.py#L10). Setting it to 0 [like #126 does](https://github.com/RobotLocomotion/drake-ros/pull/126/files#diff-c8bb268b7fab7c585431498247d1dd542eb2666de17b607dc9a1b4fd9a7c8111R163) causes the `PG * PI` term in the equation [`B+PG*PI+10` to go to zero](https://cyclonedds.io/docs/cyclonedds/0.9.1/config.html#controlling-port-numbers). Each DDSI participant is supposed to have 2 unique unicast ports, but all of them would pick the same two ports `B+10` and `B+11`. I think that's why `ParticipantGain` is needed.

gbiggs, 2022-09-11:
> I am a bit confused about when multicast or unicast is used. Are there ROS-specific docs about this? (may relate https://github.com/RobotLocomotion/drake-ros/issues/103)

My understanding is that multicast is used by default for:

- Initial "I'm here, is anyone out there?" broadcast on startup and at a fixed interval (8 seconds, I think?) thereafter
- Exchange of participant information
- Sending of data in some cases (probably when there is more than one destination)

> Should this be mentioned to some extent in ROS documentation?

Yes, and it will be for Iron Irwini. As part of improving the out-of-the-box DDS experience for ROS 2 users we're going to be revising how discovery is performed and when multicast is used, and documenting all of it.

EricCousineau-TRI, 2022-09-12:
> It looks like ParticipantGain [...]

Ah, sorry, maybe  I worded my question poorly. I understand why `ParticipantGain > 0` when needing to rely on unicast facets.

My question is moreso:
Is there a way to completely disable unicast traffic (discovery) facets and only rely on multicast traffic for Cyclone DDS, failing fast if we cannot multicast, such that we never hit participant count limits?

Specifically, how do we only ever use this feature from Cyclone DDS docs?
https://cyclonedds.io/docs/cyclonedds/0.9.1/config.html#controlling-port-numbers
> When it is none (which is the default) it simply ignores the ‘participant index’ altogether and asks the kernel to pick random ports (>= 32768). This eliminates the limit on the number of standalone deployments on a single machine and works just fine with multicast discovery while complying with all other parts of the specification for interoperability. However, it is incompatible with unicast discovery.

---

> I don't know of any ROS specific documentation [...]
> Yes, and it will be for Iron Irwini [...]

Sweet! That being said, if users can configure their setup to remove participant limitations.

sloretz, 2022-09-12:
> Is there a way to completely disable unicast traffic (discovery) facets and only rely on multicast traffic for Cyclone DDS, failing fast if we cannot multicast, such that we never hit participant count limits?

I don't see a config option to make CycloneDDS fail when unicast discovery is disabled and multicast discovery is not possible. Setting [`ParticpantIndex` to `none` disables unicast discovery](https://github.com/RobotLocomotion/drake-ros/blob/a083baf4fcdd508727c217aefee31f7814832bc7/bazel_ros2_rules/ros2/resources/rmw_isolation/rmw_isolation.py#L153-L154), but the result when `ROS_LOCALHOST_ONLY` is set and multicast is disabled is just a runtime warning: `selected interface "lo" is not multicast-capable: disabling multicast"`.

A user may want to start multiple processes that don't communicate with each other (like unit tests), so it seems like a valid configuration to me. I think we'd have to add the check in a shim in Anzu. Maybe a shim could check if `ROS_LOCALHOST_ONLY` is set, and if so terminate if the loopback interface doesn't have the multicast flag set?

EricCousineau-TRI, 2022-11-28:
Action item for me is to reexamine workflows and possibly weigh against #183

EricCousineau-TRI, 2023-02-17:
Closing in lieu of #183

closed 2023-02-17 by EricCousineau-TRI