RobotLocomotion/drake-ros #142: Should add option to visualize simulation rate in RViz?

EricCousineau-TRI, 2022-09-22:
Motivated by https://github.com/robotlocomotion/drake-ros/pull/133#pullrequestreview-1117001491

In Drake Visualizer, we have the option to show simulation rate.
I believe this is also the case for `meldis` / MeshCat.

Is there a way that we could a text overlay or even an updating text item in the "Dipslays" where we can show text?
From looking at currently available plugins, I don't see anything obvious for "put text on screen".
(this is on Jammy + Humble, using https://github.com/sloretz/apptainer-ros/blob/ad1867258b5e8a82a72a47e0eb572e71cc407093/jammy-ros-humble-desktop/Apptainer)

\cc @IanTheEngineer @sloretz

adeeb10abbas, 2022-11-01:
I think [rviz_2d_overlay_plugins](https://github.com/teamspatzenhirn/rviz_2d_overlay_plugins) would be ideal for this. They *recently* ported their ROS1 plugin to ROS2

EricCousineau-TRI, 2022-11-20:
Ah, nice! Since they have their stuff on [ROS Index](https://index.ros.org/p/rviz_2d_overlay_plugins/github-teamspatzenhirn-rviz_2d_overlay_plugins/#humble), it should be easy to add that in for CMake / Bazel testing when using debs.

adeeb10abbas, 2023-01-19:
fwiw - I tried using it. Got stuck here, posted the [issue on the original repo](https://github.com/teamspatzenhirn/rviz_2d_overlay_plugins/issues/15)