RobotLocomotion/drake-ros #151: cc, py: Should avoid ODR and recompilation for cc targets linking against static Drake targets

EricCousineau-TRI, 2022-10-17:
Regarding discussion today, the following has example of ODR:
https://github.com/EricCousineau-TRI/repro/tree/1bc74a62/bazel/bazel_shared_lib_odr_instrumentation

This is akin to solution employed in Anzu (prototyped in Anzu PR 9421):
https://github.com/EricCousineau-TRI/repro/tree/1bc74a62/drake_stuff/drake_downstream_pybind
Note that this does a bit that what is present Anzu master, as evidenced by unchecked boxes in PR 9421.

FYI @sloretz @IanTheEngineer

sloretz, 2023-01-06:
Do I understand correctly that you want `drake_ros_*` to have both static libraries and shared libraries in the same build, but you want the shared libraries to use the object files from the static libraries instead of compiling the source files again?

Reading the [`bazel_shared_lib_odr_instrumentation`](https://github.com/EricCousineau-TRI/repro/tree/1bc74a62/bazel/bazel_shared_lib_odr_instrumentation) example, it seems like it has static libraries ex2_a, ex2_b, ex2_c, and then shared library ex2_a_shared plus different ways of creating shared libraries for b and c. The "good" way in the example is to get some of the object files from `CcInfo` from the static b and c targets, then create shared libraries for b and c that use those object files as their sources plus a dependency on ex2_a_shared. I see how that allows shared library targets for `b` and `c` without recompiling `b` and `c`'s sources, but I would have assumed the object files from the b and c static libraries are not compiled with `-fPIC`. Wouldn't the files need to be recompiled anyways with `-fPIC` so the shared library code loaded anywhere in memory?

EricCousineau-TRI, 2023-01-10:
> [...] but you want the shared libraries to use the object files from the static libraries instead of compiling the source files again?

Yup!

> Wouldn't the files need to be recompiled anyways with -fPIC so the shared library code loaded anywhere in memory?

Yup! In Drake (and Anzu), we already use `build --force_pic`:
https://bazel.build/reference/command-line-reference#flag--force_pic
https://github.com/RobotLocomotion/drake/blob/v1.11.0/tools/bazel.rc#L13

That being said, it seems like `-fPIC` seems to always appear there at least for the couple of things that I tried
https://github.com/EricCousineau-TRI/repro/commit/1f03ded043481760617a537d0848c42ea256b14e

Not sure why that happens... but that being said, I think we should at least reflect those three flags from Drake / Anzu
Could you be sure to add those in PRs that'd be handling this?

EricCousineau-TRI, 2023-02-28:
There are some of the following solution options:

1. Simply recompile into `shared_library` targets, as is done per `drake-external-examples: https://github.com/RobotLocomotion/drake-external-examples/blob/909101ae7f39/drake_bazel_external/apps/BUILD.bazel#L82-L101
  - We used to do this in Anzu, but then upgraded to utility mentioned above. See also example in `dairlib`: 
2. Use simplified relinking step, as was done before in Anzu, and was the predecessor to above utility
3. Use more complex analysis and relinking, either by patching `drake-ros` BUILD file from Anzu, or incorporating it here.

I will try (2) briefly, and revert to (1) if that escapes my ~30min timebox.

jwnimmer-tri, 2023-02-28:
Maybe I should file a different issue, but it might save you some time here to fix sooner than later --

The "core" and "tf2" and "viz" libraries are _tiny_ and have roughly the same "deps" across all three.  I think trying to ship micro-libraries from drake-ros will cause more packaging trouble over time than any value it adds.  I think it should all be just 1 static library, and 1 shared library, and 1 python module (using the 1 shared library).  That will reduce your packaging headaches tenfold over the next couple years.

PR #239 (2023-03-07): [py] Avoid ODR violations in bindings
Resolves #151

Requires #238

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake-ros/239)
closed 2023-03-07 by EricCousineau-TRI