RobotLocomotion/drake-ros #157: ament: Should ensure pydrake is available on PYTHON path

EricCousineau-TRI, 2022-10-19:
Per #156, I seemed to need to add something like `export PYTHONPATH=/opt/drake/lib/python3.10/site-packages` to make `colcon test` pass locally.

We should fix this.

Possible solutions:
- Similar to #143 via ament hook.
- Adding ctest env vars, similar to https://github.com/RobotLocomotion/drake-external-examples/pull/155

@cottsay Can you handle this over next two weeks?
Ideally, we don't have to figure out Python major/minor. I'm not sure if Drake's CMake setup has affordance for it; if not, ideally ament / ros2 has some affordance.

\cc @IanTheEngineer @sloretz

cottsay, 2022-10-20:
> Can you handle this over next two weeks?

Sure, I'll prioritize this.

Here is where the hook is defined for `ament_cmake_python` packages: https://github.com/ament/ament_cmake/blob/rolling/ament_cmake_python/ament_cmake_python-extras.cmake

All of those CMake macros are "private", so we'll either need to promote some or all of it to public so that we can call it here, or we'll need to duplicate the logic. I wasn't able to find anything in the CMake that Drake provides that would give us what we need.

Question: Drake's stance is that everybody should just use RPATHs to avoid the need for LD_LIBRARY_PATHs. That isn't appropriate in the context of ROS, so we needed to add our own hooks to deal with it. That said, what is Drake's "recommended" way for downstream software to discover pydrake?

jwnimmer-tri, 2022-10-20:
> Drake's stance is that everybody should just use RPATHs to avoid the need for LD_LIBRARY_PATHs.

That's not correct.  Users who are compiling their C++ code against Drake are invited to use any linker/loader mechanisms they choose.  We don't prefer one over another, it's just a question of the user's own preference for their build.  The loading mechanism is property of the users _own_ executables.

> What is Drake's "recommended" way for downstream software to discover pydrake?

We don't expect anything special for Python.  Any of the standard mechanisms should work fine, e.g., adding to `PYTHONPATH` or using a virtual environment with `pip install`, etc.

cottsay, 2022-10-20:
> That's not correct. Users who are compiling their C++ code against Drake are invited to use any linker/loader mechanisms they choose. We don't prefer one over another, it's just a question of the [...]

You're right to call me out on this. That assertion is wrong and comes off as accusatory, which is not how I meant it. I feel the need to explain myself.
Dropping `LD_LIBRARY_PATH` from the Dockerfile regressed the runtime discoverability of `libdrake.so` independently of the build-time discovery using CMake. Under `robotlocomotion/drake:jammy`, a "vanilla" `CMakeLists.txt` is able to `find_package(drake)` just fine using the modification to the `PATH` variable baked into the image. By default, an executable then linked against `libdrake.so` will have an RPATH when it is built, but the RPATH is stripped when installed (again, by default).
While my comment wasn't fair by any measure, the decision to remove LD_LIBRARY_PATH does imply that downstream projects must either specifically disable the stripping of RPATHs, or specifically modify LD_LIBRARY_PATH for built and installed executables to be run-able. The only reason I found this to be unexpected is that I don't need to do anything special to enable CMake to find drake at build-time - only once my project is installed does the problem surface.

---

All that said, this ticket is about `PYTHONPATH`!

The Dockerfile is already setting it here: https://github.com/svenevs/drake/blob/56d36734bca4540dc1ef4c31535a3f3429c3d3c5/tools/install/dockerhub/jammy/Dockerfile#L14-L15

@EricCousineau-TRI, under what environment do you find that you need to set it manually? Do you also need to modify your environment for CMake to find drake?

EricCousineau-TRI, 2022-10-20:
My consumption mode is via Drake `*.tar`. I'd prefer that we ensure this workflow is functional w/ no hand-holding, and we remove the hack from the Dockerfile.

From the example in the overview: https://github.com/RobotLocomotion/drake-external-examples/pull/155
It's set using the following (kind-of fragile) mechanism:
https://github.com/RobotLocomotion/drake-external-examples/blob/eabc1a36bd7fe1537aadbb559e6af76ed3878d9f/drake_cmake_external/drake_external_examples/CMakeLists.txt#L38-L50

EricCousineau-TRI, 2022-10-20:
Filed drake#18150, but for now, I think we can still go the `drake-external-examples` route.

PR #160 (2022-11-01): Add Ament hook to put pydrake in PYTHONPATH
~~This change borrows from the `PYTHONPATH` hook creation mechanism used in `ament_cmake_core`.~~

~~It makes the following assumptions:~~
~~1. Drake installs Python modules to `platlib`, and not `purelib`~~
~~2. Drake installs Python modules to the same prefix as `libdrake.so`~~
~~3. Drake uses the same Python interpreter version as the system's default~~

~~Closes #157~~
~~Related to #143~~

Requires RobotLocomotion/drake#18201

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake-ros/160)
closed 2022-11-01 by EricCousineau-TRI