RobotLocomotion/drake-ros #159: conversions: Additional conversions to provide

EricCousineau-TRI, 2022-10-20:
We should either provide these here, or as a package in this workspace. 

We should provide 

- [ ] Python versions of `geometry_conversions.h` (either bindings, or reimplementations)
- [ ] Eigen matrices
- [ ] Point Clouds (\cc @hongkai-dai)
- [ ] Images

adityapande-1995, 2023-01-28:
Made an upstream ticket for ROS messages to be typecastable from python to C++ : https://github.com/ros2/rosidl/issues/723

adityapande-1995, 2023-02-13:
https://github.com/RobotLocomotion/drake-ros/pull/215 Adds bindings for ``geometry_conversions.h`` and Eigen matrices to numpy.

adityapande-1995, 2023-04-04:
@EricCousineau-TRI For pointclouds, IIUC, we need a conversion between the python and C++ ros message type (using something more efficient then serializing-deserializing) and then between the C++ pointcloud message and  drake C++ type, right ?