RobotLocomotion/drake-ros #172: ament env hooks: Paths are prepended before workspace values?

EricCousineau-TRI, 2022-10-21:
When testing #160, I noticed the following on `main` at 63a110a
```
$ rm -rf build/ install/ log/
$ colcon build --packages-up-to drake_ros_core
$ bash-isolate
$ source ./install/setup.sh
$ echo $LD_LIBRARY_PATH | sed -e 's#:#\n#g' -e 's#'${PWD}'#${PWD}#g'
/opt/drake/lib
${PWD}/install/drake_ros_core/lib
/opt/ros/humble/opt/rviz_ogre_vendor/lib
/opt/ros/humble/lib/x86_64-linux-gnu
/opt/ros/humble/lib
```

Note how `/opt/drake/lib` precedes workspace's lib. We should ideally insert workspace after this, to admit (un)intended shadowing.

@cottsay Would you be able to do this in ~2 weeks?
I assume there's some configuration for the env hooks to say "eval this before primary workspace env var manipulations."

\cc @IanTheEngineer @sloretz

sloretz, 2023-01-06:
I think this could be done by changing [`prepend-non-duplicate`](https://github.com/cottsay/drake-ros/blob/63a110a6d6648edc7e1cb755cc2ef297d80405b2/drake_ros_core/CMakeLists.txt#L73) to [`append-non-duplicate`](https://github.com/colcon/colcon-core/blob/1134f504b7f4c46e61044df43912f20c99348254/colcon_core/shell/dsv.py#L63) and [`ament_prepend_unique_value`](https://github.com/cottsay/drake-ros/blob/63a110a6d6648edc7e1cb755cc2ef297d80405b2/drake_ros_core/env-hooks/drake_library_path.sh.in#L5) to [`ament_append_unique_value`](https://github.com/ament/ament_package/blob/ff6bebb0f00add20d0776ad1ef586565ef8b932c/ament_package/template/package_level/local_setup.sh.in#L47).

EricCousineau-TRI, 2023-01-09:
Ooh, yes, that'd be great! Is this something easy to do in next week?