RobotLocomotion/drake-ros #174: dev: Should focus on Bazel as primary developer workflow

EricCousineau-TRI, 2022-10-21:
While `colcon`, `ament`, etc. are nice tools, they require some overhead to learn, and I fear what this will entail once Drake Developers become more involved.

Per Drake Slack (private TRI OSRC channel, on 2022-10-18:

> In `drake-ros`, I'm not loving writing unittests against Drake's CMake interface and having to also deal w/ CMake.
> There are other facets, such as not having `rmw_isolation` readily available in CMake (which could be solved in various ways).
> 
> Solution routes:
> (a) Ensure testing utilities have parity (e.g. expose or hoist `//drake/common:eigen_matrix_compare` as installable artifact, expose `rmw_isolation`, etc.)
> (b) Do primary testing in Bazel. Do very minimal "does it compile and kinda run" testing in CMake.
> 
> My preference is for (b), as it less of an overhead for me as a dev and reviewer.

From Slack discussion w/ @IanTheEngineer, I believe we should do (b) as a means for primary development workfow

However, that does not mean we should cripple / remove the CMake workflow, for obvious reasons (`drake-ros`).
But I believe CMake could be delegated to very basic testing (e.g. "can I import this? do I get dload errros? etc."), and Bazel would be used for more advanced testing (`rmw_isolation`, running multiple nodes in a mulitprocess fashion, etc.).

@IanTheEngineer @sloretz I'd like to discuss this in next meeting for timeline.
Maybe we switch to this in Jan 2023 or so.

EricCousineau-TRI, 2022-10-21:
Relates pain points encountered in:
- #173 
- #172
- #157
- #156

sloretz, 2023-01-04:
#205 might be related to this ticket. It moves files around to make things more pleasant to do in bazel.