RobotLocomotion/drake-ros #183: ROS 2: Improve the discovery methods used by nodes to prevent conflicts and accidental discoveries

gbiggs, 2022-11-01:
We need to address a couple of common requests/complaints regarding DDS behaviour in ROS 2:

- nodes can too-easily discover other nodes that are not part of their application, leading to, for example, robots suddenly moving; and
- discovery traffic by default goes everywhere, flooding networks, which can bring down networks or at least degrade network performance significantly.

To make control of discovery both easier and more fine-grained, in Iron Irwini we are considering doing the following.

- Deprecating the ROS_LOCALHOST_ONLY environment variable.
- Adding a new environment variable, ROS_AUTOMATIC_DISCOVERY_RANGE, which will control how far discovery traffic for the automatic discovery process can travel.
- Adding a new environment variable, ROS_STATIC_PEERS, which will allow users to specify the address(es) of other hosts (robots, etc.) to connect to.

The ROS_AUTOMATIC_DISCOVERY_RANGE variable will accept the following values.

- 1: Disable automatic discovery. No other nodes will be found automatically, even on the same host.
- 2: Limit automatic discovery to the local host only. Nodes on the same host will be automatically found, unless they disable their own automatic discovery, but no discovery traffic will leave the host.
- 3: Enable automatic discovery on the local subnet. Nodes on other hosts that are on the same subnet will be found automatically, unless they disable their own automatic discovery.

The ROS_STATIC_PEERS variable will accept a semi-colon-separated list of addresses (IP addresses and hostnames searchable via DNS). All nodes on the hosts specified in this variable will be discovered, even if they have turned off automatic discovery.

The combination of these two variables will enable, for example, automatically discovering only nodes on the local host, while also talking to a robot at a known address, but nothing else on the network.

For more details and discussion, see [this Discourse thread](https://discourse.ros.org/t/proposed-changes-to-how-ros-performs-discovery-of-nodes/27640).

EricCousineau-TRI, 2022-11-07:
Thanks! Posted initial impression and requests here:
https://discourse.ros.org/t/proposed-changes-to-how-ros-performs-discovery-of-nodes/27640/17?u=ericcousineau-tri

IanTheEngineer, 2022-11-21:
Agreed on removing the text from this ticket. However, it would be nice to see a meta-ticket on one of the relevant repositories that describes the different areas of improvement (preferably with check-boxes) to serve as a central point to link against when its eventually time to create PR's.

EricCousineau-TRI, 2022-12-19:
Centralized issue: https://github.com/ros2/ros2/issues/1359

mxgrey, 2023-02-06:
Note that there has been significant progress in making a testbed for the behaviors in this PR: https://github.com/osrf/iron_irwini_discovery_modes_prototype/pull/2

mxgrey, 2023-03-13:
Relevant PRs:
* https://github.com/ros2/rmw/pull/349
* https://github.com/ros2/rcl/pull/1038
* https://github.com/ros2/rmw_fastrtps/pull/667
* https://github.com/ros2/rmw_cyclonedds/pull/441
* https://github.com/ros2/rmw_cyclonedds/pull/429
* https://github.com/ros2/rmw_connextdds/pull/108
* https://github.com/ros2/system_tests/pull/512

EricCousineau-TRI, 2023-04-10:
Closed per our TRI SoW

closed 2023-04-10 by EricCousineau-TRI