RobotLocomotion/drake-ros #214: Should support ros2 launch

EricCousineau-TRI, 2023-01-20:
Most likely would happen in ~August, pending an external solution and/or upped priority

\cc @adeeb10abbas

EricCousineau-TRI, 2023-01-20:
Should also look at alternative Bazel + ros2 approaches, and how they do it:
https://github.com/RobotLocomotion/drake-ros/tree/4c46abbfc5566c9b4cd1c3dffb77ec6549aa3a21/bazel_ros2_rules/ros2#alternatives

sloretz, 2023-02-22:
How should Launch be supported? Is this looking for a bazel rule that invokes a launch file, kind of like a `*_binary()` rule?

```bazel
# launch with `bazel run //:robot_launch`
ros2_launch_binary(
   name = "robot_launch",
   launch_file = ":robot_launch.xml",
)
```

EricCousineau-TRI, 2023-02-24:
Yup!

Though I wouldn't say this is a priority for Anzu usage yet, though.

adityapande-1995, 2023-07-27:
For starters, I feel the CLI should also be supported, as that is the easiest way to use ros2 launch.
For e.g. : 

```bazel run @ros2//:ros2 launch some_launch_file.py```

I get an error saying : 
```
    for entry_point in distribution(dist_name).entry_points
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 957, in distribution
    return Distribution.from_name(distribution_name)
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 548, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: No package metadata was found for ros2cli
```

The underlying ros2 archive extract which contains the run.bash hack actually does work. If you run the above command with ``--sandbox_debug`` and ``cd`` into the cached archive, run : 
```
./run.bash ros2 launch some_launch_file.py
```

this works perfectly fine.

We're missing data dependencies or environment variables in ``_ros2_shim.py``. Investigating.

EricCousineau-TRI, 2023-07-27:
FWIW,  we ran into this in Anzu.

I'm a bit murky on exact details, but I now think the `No package metadata was found for ros2cli` error will go away if you refetch the repository (e.g. `bazel clean --expunge --async`, or change anything in the repository definition to trigger re-fetching).

adityapande-1995, 2023-08-14:
In other news, running ``bazel run @ros2//:ros2`` from inside ``ros2_examples_bazel_installed`` actually works without complaining about a ros2cli dependency. If we do add a rule to run launch tests, it should run consistently from inside ``drake_ros`` as well, ideally.

EricCousineau-TRI, 2024-04-18:
Resolved by #298

FYI @tervay-bdai @adeeb10abbas @ali-bdai

EricCousineau-TRI, 2024-04-18:
Follow-on work documented in #344