RobotLocomotion/drake-ros #221: [ci] Failure in bazel_build_and_test

EricCousineau-TRI, 2023-02-16:
Seen as part of #210

https://github.com/RobotLocomotion/drake-ros/actions/runs/4195219820/jobs/7274451434#logs

```sh
Invoking "sudo apt-get upgrade -y"
...
  Setting up grub-efi-amd64-signed (1.187.3~20.04.1+2.06-2ubuntu14.1) ...
  mount: /var/lib/grub/esp: special device /dev/disk/by-id/scsi-14d5346542020202076e4e0de6b21a348aa1740149647cee7-part15 does not exist.
  dpkg: error processing package grub-efi-amd64-signed (--configure):
   installed grub-efi-amd64-signed package post-installation script subprocess returned error exit status 32
...
   Errors were encountered while processing:
   grub-efi-amd64-signed
  E: Sub-process /usr/bin/dpkg returned an error code (1)
```

@sloretz Is this something you could fix (directly or via delegation) over next few days?

\cc @adityapande-1995

jwnimmer-tri, 2023-02-16:
FYI in case it helps, in Anzu CI we do this on our Ubuntu builders to prevent useless (and sometimes broken) upgrade attempts of kernels (which are moot for ephemeral build containers anyway):

```
echo 'APT::Get::Assume-Yes "true";' | sudo tee /etc/apt/apt.conf.d/90yes
echo "debconf debconf/frontend select Noninteractive" | sudo debconf-set-selections
sudo apt-mark hold grub-pc-bin linux-aws linux-headers-aws linux-image-aws
```

That's on EC2, not GitHub Actions, but the idea might apply elsewhere.

EricCousineau-TRI, 2023-02-16:
Thanks! I'll give that a quick whirl.

EricCousineau-TRI, 2023-02-16:
Had filed #224, but ran into some misconfiguration on my side.

However, this seems like an infrastructure flake (or something else), as the error mode has changed. Now, Bazel fails on `lldb` as of d88cb47
https://github.com/RobotLocomotion/drake-ros/actions/runs/4198834649/jobs/7282965215
```
 //ros2_example_apps:lldb_oracle_cc_test                                  FAILED in 0.5s
```

Investigating...

EricCousineau-TRI, 2023-02-17:
Leveraging #226, this is what I see on CI:
```sh
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'lldb.embedded_interpreter'
(lldb) target create "ros2_example_apps/oracle_cc"
Current executable set to '{runfiles}/ros2_example_bazel_installed/ros2_example_apps/oracle_cc' (x86_64).
(lldb) settings set -- target.run-args  "--ros-args" "--disable-external-lib-logs"
(lldb) command source -s 0 'ros2_example_apps/test/oracle_cc.lldb'
Executing commands in '{runfiles}/ros2_example_bazel_installed/ros2_example_apps/test/oracle_cc.lldb'.
(lldb) settings set target.process.stop-on-exec false
(lldb) process launch --stop-at-entry
error: 'A' packet returned an error: 8
```

In contrast, this is what I see on my machine as of 
```sh
$ bazel test --nocache_test_results --test_output=streamed //ros2_example_apps:lldb_oracle_cc_test
...
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'lldb.embedded_interpreter'
(lldb) target create "ros2_example_apps/oracle_cc"
Current executable set to '{runfiles}/ros2_example_bazel_installed/ros2_example_apps/oracle_cc' (x86_64).
(lldb) settings set -- target.run-args  "--ros-args" "--disable-external-lib-logs"
(lldb) command source -s 0 'ros2_example_apps/test/oracle_cc.lldb'
Executing commands in '{runfiles}/ros2_example_bazel_installed/ros2_example_apps/test/oracle_cc.lldb'.
(lldb) settings set target.process.stop-on-exec false
(lldb) process launch --stop-at-entry
Process 30 launched: '{runfiles}/ros2_example_bazel_installed/ros2_example_apps/oracle_cc' (x86_64)
(lldb) breakpoint set --method 'Oracle::publish_status'
Breakpoint 1: no locations (pending).
WARNING:  Unable to resolve breakpoint to any actual locations.
(lldb) continue
1 location added to breakpoint 1
Process 30 resuming
Process 30 stopped
* thread #1, name = '_oracle_cc_nosh', stop reason = breakpoint 1.1
    frame #0: 0x0000555555584770 _oracle_cc_noshim`ros2_example_apps::Oracle::publish_status()
_oracle_cc_noshim`ros2_example_apps::Oracle::publish_status:
<good stuff>
<success>

```

EricCousineau-TRI, 2023-02-17:
Most likely fixable using suggestions in trace of posts linked from https://stackoverflow.com/a/73613377/7829525

EricCousineau-TRI, 2023-02-17:
#227 allows it to work:
https://github.com/RobotLocomotion/drake-ros/actions/runs/4199377731/jobs/7284169136#step:10:61
```
//ros2_example_apps:lldb_oracle_cc_test                                  PASSED in 4.5s
```

EricCousineau-TRI, 2023-02-23:
Per comment, reopening: https://github.com/RobotLocomotion/drake-ros/pull/224#event-8542374645

PR #227 (2023-02-17): [ci] Ensure we can use lldb on Docker for GitHub actions
Resolves #221

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake-ros/227)
closed 2023-02-23 by sloretz