RobotLocomotion/drake-ros #230: installation instructions in drake_ros_examples/README.md doesn't obtain `colcon`?

RussTedrake, 2023-02-20:
Unless I missed something. 

I had to manually install it following instructions from https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.html

adeeb10abbas, 2023-02-20:
I don't think you missed anything. I think the installation instructions don't include either `rosdep` or `colcon` it's just assumed the user would have it... although it would be very useful

IanTheEngineer, 2023-02-20:
Thanks for opening this issue, Russ. The instructions for getting set up are in the `drake_ros_core` README and lean heavily on the ROS2 setup instructions:
https://github.com/RobotLocomotion/drake-ros/blob/main/drake_ros_core/README.md

That being said, I ran through the instructions, and the default ROS Humble / Rolling installation does not have the user installing `colcon` or `rosdep` for whatever reason. To me, that means our setup guide should  be more self contained. Simultaneously, we'll speak with folks to improve the upstream ROS2 setup docs.

EricCousineau-TRI, 2023-02-20:
Thanks! We should address this after we land #205, just so we have a more cohesive and targeted location for concise, non-redundant instructions.

EricCousineau-TRI, 2023-02-27:
I believe this should be resolved by #235. Let us know if it's still not clear.

closed 2023-02-27 by EricCousineau-TRI