RobotLocomotion/drake-ros #231: ... "ament_cmake_clang_format", but CMake did not find one.

RussTedrake, 2023-02-20:
on a relatively fresh jammy install, running the installation instructions in `drake_ros_examples/README.md`, I get
```
russt@home-cart13:~/tmp/qr_ros$ colcon build
Starting >>> drake_ros_core
--- stderr: drake_ros_core                         
CMake Error at CMakeLists.txt:122 (find_package):
  By not providing "Findament_cmake_clang_format.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "ament_cmake_clang_format", but CMake did not find one.

  Could not find a package configuration file provided by
  "ament_cmake_clang_format" with any of the following names:

    ament_cmake_clang_formatConfig.cmake
    ament_cmake_clang_format-config.cmake

  Add the installation prefix of "ament_cmake_clang_format" to
  CMAKE_PREFIX_PATH or set "ament_cmake_clang_format_DIR" to a directory
  containing one of the above files.  If "ament_cmake_clang_format" provides
  a separate development package or SDK, be sure it has been installed.

---
Failed   <<< drake_ros_core [1.17s, exited with code 1]

Summary: 0 packages finished [1.35s]
  1 package failed: drake_ros_core
  1 package had stderr output: drake_ros_core
```

Seems related to #88.

RussTedrake, 2023-02-20:
Ah.  it's possible that it's because i'm still on humble.  Will try rolling.

RussTedrake, 2023-02-20:
Resolved on my local machine via
```
sudo apt install ros-rolling-ament-cmake-clang-format ros-rolling-ament-cmake-pycodestyle ros-rolling-test-msgs

```

IanTheEngineer, 2023-02-20:
Thanks for reporting this, Russ. similar to #230, this points to our setup documentation needing more clarity. I'll take a pass at updating the initial setup instructions.

EricCousineau-TRI, 2023-02-20:
Similar to #230, we should address this after we land #205.

EricCousineau-TRI, 2023-03-13:
Resolved by ensuring `rosdep install` is called, with correct `package.xml`

closed 2023-03-13 by EricCousineau-TRI