RobotLocomotion/drake-ros #237: [build] Consider offering only one library

EricCousineau-TRI, 2023-02-28:
Rather than offering multiple C++ and Python libraries, we may just want to consolidate.

See @jwnimmer-tri's comment here:
https://github.com/RobotLocomotion/drake-ros/issues/151#issuecomment-1448484239

I think we can do this within the next ~2 months or so.

\cc @sloretz If we can get to this by end of this month, then great! But I would pin this at lower priority than resolving our outstanding work first!

jwnimmer-tri, 2023-02-28:
I also just looked at the python example:

https://github.com/RobotLocomotion/drake-ros/blob/94a32508bb7e1e1cfb5c422ac7a64bc836d9dd3d/drake_ros_examples/examples/iiwa_manipulator/iiwa_manipulator.py#L18-L20

Those imports don't make any sense.  We don't want to have 3 different packages on PyPI -- _that_ will be even worse ODR hazards (we'd need to version-lock them all to each other).

We're going to need to switch up everyone's import paths.  The sooner we can do that, the fewer users will experience churn.

EricCousineau-TRI, 2023-02-28:
Yup, I had actually already started towards resolving #151.

jwnimmer-tri, 2023-03-14:
Any further thoughts on nuking the sub-namespaces / sub-modules as well (`core`, `tf2`, ...)?
https://github.com/RobotLocomotion/drake-ros/issues/151#issuecomment-1448484239

I haven't tried out the idea as a user yet, but my intuition is that having a single module would be the simplest for users to import and use, and for BUILD maintenance moving forward.  There are only ~3 components per module currently, and their dependencies don't seem to differ that much.

I could work on a draft PR if that helps assess.

EricCousineau-TRI, 2023-03-14:
A draft PR would be excellent - I feel biased towards sub-namespaces and sub-modules, but I can be convinced otherwise!