RobotLocomotion/drake-ros #243: [bazel] Should ensure pytest is actually run

EricCousineau-TRI, 2023-03-03:
It's unclear to me if any `pytest`-based code is getting run via our Bazel setup.

My suggested options:
- Figure out how to have `pytest` run explicitly when `__name__ == "__main__"`
- Use `unittest` instead (my preference)

\cc @sloretz

sloretz, 2023-03-03:
See https://github.com/EricCousineau-TRI/drake-ros/pull/6 which invokes pytest via

```python
if __name__ == '__main__':
    sys.exit(pytest.main(sys.argv))
```

> Use unittest instead (my preference)

One blocker is `ament_add_test` expects tests to output a junit xml file, which unittest doesn't do by default. Maybe this package would do it though? https://packages.ubuntu.com/jammy/python3-xmlrunner

jwnimmer-tri, 2023-03-03:
Yup.  That's what [Drake uses](https://github.com/RobotLocomotion/drake/blob/v1.13.0/tools/workspace/xmlrunner_py/repository.bzl) for its python unit tests.

EricCousineau-TRI, 2023-03-06:
Adding `pytest.main` works for me for now, and will resolve it.

FWIW, I don't love the "implicitness" of `pytest` (plugins, configuration, etc), and generally prefer `unittest`, if just for more clarity on what's happening. However, I don't want to sacrifice other work for this.

EricCousineau-TRI, 2023-03-08:
I think this has been resolved-ish by ensuring we use the `pytest` ending stanza.

closed 2023-03-08 by EricCousineau-TRI