RobotLocomotion/drake-ros #244: License for examples

jwnimmer-tri, 2023-03-03:
For `ros2_example_bazel_installed` and other examples, I recommend that we switch to [MIT-0](https://spdx.org/licenses/MIT-0.html) as the license for those directories.

That's what we're doing in [drake-external-examples](https://github.com/RobotLocomotion/drake-external-examples/issues/240).

I think we want users to copy some or all parts of the examples into their own projects to get started.  As it stands today, formally they need to copy the Apache-2.0 boilerplate as well, since the condition of forking our code is that they preserve the notices.  With MIT-0, they can remove the notices when they fork, and use their own license.

EricCousineau-TRI, 2023-03-06:
SGTM.
In #246, I'm removing the OSRF license headers per out-of-band discussion.
After that, I'd be fine with relicensing for MIT-0.

Out of curiosity, have you seen linting + fixing tooling that can enforce this for us?
I guess I don't love the the manual aspect of affixing license headers, and would love if we didn't have to do this, or at a minimum, had tooling slap it in for us.

jwnimmer-tri, 2023-03-06:
To be clear:
- I think library software (e.g., `drake_ros/**`) should remain as `Apache-2.0`.
- I think sample code (`drake_ros_examples/**`) should switch to `MIT-0`.

For linting, cpplint can check `Copyright [year] <Copyright Owner>` but I think that's not really what you're looking for.  I don't know of anything else, but it's not too bad to write a bespoke one on Python.  We have a few similar things sitting around in Drake, e.g., `drake/tutorials/notebook_lint_test.py`.

If you like, peek at https://github.com/RobotLocomotion/drake/blob/master/examples/hardware_sim/hardware_sim.py in Drake.  That's one stab at a 2-line boilerplate.  Per https://github.com/RobotLocomotion/drake/issues/1805#issuecomment-550411900, it's probably worthwhile to have at least a short snippet in every file.  I agree that the dozen lines you're nuking are annoying, but to me the two-line format is okay.

See also https://spdx.dev/ids/.

EricCousineau-TRI, 2023-03-06:
Sweet, that all sounds great, and yeah, I'd be happy w/ 2 liners to preserve local info.

jwnimmer-tri, 2023-03-14:
I guess I'll put together a PR.

jwnimmer-tri, 2024-07-09:
I don't anticipate anyone will attend to this issue, so I'll close it as "not planned".

closed 2024-07-09 by jwnimmer-tri