RobotLocomotion/drake-ros #247: [ci] Should improve bazel caching

EricCousineau-TRI, 2023-03-07:
See https://drakedevelopers.slack.com/archives/C2CP2E50A/p1673880604846249

jwnimmer-tri, 2023-03-22:
I'm pretty sure that the caching is all wrong.

By default, GitHub Actions only _write back_ to the cache when the `key:` string changes.  Since our key string is a literal with no substitutions, it is never written back, so it's getting progressively colder as we continue to change the code.

See https://github.com/RobotLocomotion/drake-blender/pull/7 where I'm iterating to get this right.  I'll circle back here once I've tested it there on a smaller repo.

EricCousineau-TRI, 2023-03-22:
Ah, good to know, and thanks!

EricCousineau-TRI, 2023-09-25:
Getting close, but the job can still get cancelled sometimes:
https://github.com/RobotLocomotion/drake-ros/actions/runs/6303897969/job/17114097008

It seems like it is still running out of disk space:
https://github.com/RobotLocomotion/drake-ros/actions/runs/6303897969/attempts/1
```
Unhandled exception. System.IO.IOException: No space left on device : '/home/runner/runners/2.309.0/_diag/Worker_20230925-192117-utc.log'
```
@adityapande-1995 Would you be able to look into this?
\cc @calderpg-tri @IanTheEngineer 

FWIW, GitHub Workflows seems a bit wonky right now:
- I don't always see instructions for the current executing step
- If I re-run the job, then I cannot see the old logs

jwnimmer-tri, 2023-09-25:
BTW If you're running `apt` commands in the build (either update and/or upgrade), make sure you nuke their downloaded files and lists database afterwards:
https://docs.docker.com/develop/develop-images/dockerfile_best-practices/#apt-get

EricCousineau-TRI, 2023-10-30:
Resolved by #309

PR #248 (2023-03-07): [ci] Narrow bazel cache to repository and disk cache
Resolves #247

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake-ros/248)
closed 2023-10-30 by EricCousineau-TRI