RobotLocomotion/drake-ros #266: [bazel] AssertionError after ros-humble-navigation2 package is installed.

svperbeast, 2023-03-27:
Hello, 

When i run `bazel` command, i get `AssertionError`:

```bash
ERROR: error loading package under directory '': no such package '@ros2//': Failed to setup @ros2 repository: './run.bash /home/.../.cache/bazel/_bazel_.../5a9db91546a31d78cc72a3c4d250d476/external/bazel_ros2_rules/ros2/scrape_distribution.py -i action_msgs -i builtin_interfaces -i console_bridge_vendor -i rclcpp -i rclcpp_action -i rclcpp_components -i rclpy -i ros2cli -i ros2cli_common_extensions -i rosidl_default_generators -i rosidl_default_runtime -i std_msgs -i nav_msgs -i nav2_msgs -i geometry_msgs -i tf2_py -i rmw_fastrtps_cpp -o distro_metadata.json' exited with 1
--- captured stderr ---
Traceback (most recent call last):
  File "/home/.../.cache/bazel/_bazel_.../5a9db91546a31d78cc72a3c4d250d476/external/bazel_ros2_rules/ros2/scrape_distribution.py", line 54, in <module>
    main()
  File "/home/.../.cache/bazel/_bazel_.../5a9db91546a31d78cc72a3c4d250d476/external/bazel_ros2_rules/ros2/scrape_distribution.py", line 46, in main
    distro = scrape_distribution(
  File "/home/.../.cache/bazel/_bazel_.../5a9db91546a31d78cc72a3c4d250d476/external/ros2/resources/ros2bzl/scraping/__init__.py", line 91, in scrape_distribution
    index_all_packages(), include, exclude)
  File "/home/.../.cache/bazel/_bazel_.../5a9db91546a31d78cc72a3c4d250d476/external/ros2/resources/ros2bzl/scraping/__init__.py", line 26, in index_all_packages
    packages = {
  File "/home/.../.cache/bazel/_bazel_.../5a9db91546a31d78cc72a3c4d250d476/external/ros2/resources/ros2bzl/scraping/__init__.py", line 27, in <dictcomp>
    name: collect_ros_package_metadata(name, prefix)
  File "/home/.../.cache/bazel/_bazel_.../5a9db91546a31d78cc72a3c4d250d476/external/ros2/resources/ros2bzl/scraping/metadata.py", line 127, in collect_ros_package_metadata
    plugin_libraries.extend(parse_plugins_description_xml(
  File "/home/.../.cache/bazel/_bazel_.../5a9db91546a31d78cc72a3c4d250d476/external/ros2/resources/ros2bzl/scraping/metadata.py", line 37, in parse_plugins_description_xml
    assert root.tag == 'library'
AssertionError
```

https://github.com/RobotLocomotion/drake-ros/blob/70e06d8192cb5e1fca591a233ea1ea74870f53d9/bazel_ros2_rules/ros2/resources/ros2bzl/scraping/metadata.py#L34-L38

### Steps to reproduce

1. install `navigation2` package: `sudo apt install ros-humble-navigation2`
2. run bazel command: `bazel build //...` or `bazel-compdb` ...

### System Information

* OS: Ubuntu Jammy
* Arch: x86_64
* Bazel: 6.1.1
* ROS: Humble (`/opt/ros/humble`)

### Workaround

Navigation2 package has xml files that root node != "library" (libraries are children of `class_libraries` node):

```xml
<class_libraries>
    <library path="nav2_rotation_shim_controller">
        <class type="nav2_rotation_shim_controller::RotationShimController" base_class_type="nav2_core::Controller">
            <description>
                nav2_rotation_shim_controller
            </description>
        </class>
    </library>
</class_libraries>
```

The following code seems work for me, but i'm not sure that it is a correct solution:

```python
def parse_plugins_description_xml(path_to_plugins_description_xml):
    plugins_description_xml = ET.parse(path_to_plugins_description_xml)
    root = plugins_description_xml.getroot()
    if root.tag == 'library':
        return dict(plugin_libraries=[root.attrib['path']])
    else:
        plugin_libraries = []
        for library in root.findall('library'):
            plugin_libraries.append(library.attrib['path'])
        return dict(plugin_libraries=plugin_libraries)
```

Thanks!

adeeb10abbas, 2023-03-27:
Thanks for reporting this! I shared the exact same issue yesterday with Eric and today in [drake-slack](https://drakedevelopers.slack.com/archives/C2CP2E50A/p1679926642448159), didn't realize others had already run into this as well haha. 

As far as the solution goes, I think we should add more proper error reporting?

sloretz, 2023-03-27:
It looks like `pluginlib` allows either [`<library>` or `<class_libraries>` to be the top level element](https://github.com/ros/pluginlib/blob/30042370880c5a476d0a2817187aab11c1c886f6/pluginlib/include/pluginlib/class_loader_imp.hpp#L689-L700). It then looks at [all the sibling `<library>` tags](https://github.com/ros/pluginlib/blob/30042370880c5a476d0a2817187aab11c1c886f6/pluginlib/include/pluginlib/class_loader_imp.hpp#L777).

I think there are two fixable issues with `parse_plugins_description_xml()`. The first is when the root tag is `class_libraries` it should look at the child `<library>` tags, and the second is it should look at all `<library>` tags, not just the first one. I'll work on a PR with those two changes.

A nice to have would be an xsd upstream for plugin descripton xmls. I'll open an issue for that, but I may not have time to work on it.

PR #268 (2023-03-27): Fix plugin xml parsing with multiple library tags
Fixes #266

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake-ros/268)