RobotLocomotion/drake-ros #293: DRAKE_SUGGESTED_VERSION staleness

jwnimmer-tri, 2023-08-14:
The `DRAKE_SUGGESTED_VERSION` constant currently points to v1.13 (from February 15th, 2023) even though v1.19 was out in July and v1.20 is right around the corner.

Step 1 is to bump is to something newer.  I'll open a PR for that.

Step 2 is to enact a procedure to keep it pinned to the latest stable release.

How do we want do to that?  Have a GitHub bot remind us?  Or add a reminder to the https://drake.mit.edu/release_playbook.html?  Or ... ?

EricCousineau-TRI, 2023-08-15:
A reminder in the playback would be great!
\cc @IanTheEngineer @calderpg-tri

jwnimmer-tri, 2023-08-15:
@rpoyner-tri we have a proposal for a new release playbook step.  Would you be willing to try it this time around, and update the playbook with new instructions?

The request is that after Drake is tagged, the release manager opens a new issue on this repository, remaking on the need to upgrade our Drake pin here.  This is similar in spirit to the existing step of opening the Drake issue to request a Docker pin update from Betsy.

jwnimmer-tri, 2023-09-18:
Both steps are done.

closed 2023-09-18 by jwnimmer-tri