RobotLocomotion/drake-ros #327: Please update DRAKE_SUGGESTED_VERSION to v1.25.0

rpoyner-tri, 2024-01-16:
https://github.com/RobotLocomotion/drake/releases/tag/v1.25.0

jwnimmer-tri, 2024-01-23:
Apparently there are actual compilation failures with this upgrade, so `drake-ros` team please take a look ASAP:
https://stackoverflow.com/questions/77867296/colcon-build-causes-drake-errors-for-the-latest-drake-release

jwnimmer-tri, 2024-01-23:
> Apparently there are actual compilation failures with this upgrade, ...

False alarm.  The build with 1.25 works fine, the user was (I think) just using a >6 month old checkout of drake-ros.

jwnimmer-tri, 2024-01-23:
Yup, sorry for the noise!  I put too much trust in the SO post.

Since I have new the new checksum locally now anyway, I'll open a PR to see what CI says.

PR #328 (2024-01-24): [drake] Upgrade to v1.25.0
Closes #327.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake-ros/328)
closed 2024-01-24 by IanTheEngineer