RobotLocomotion/drake-ros #335: SceneMarkersSystem support for glTFs with external assets

jwnimmer-tri, 2024-02-21:
The `drake_ros::viz::SceneMarkersSystem` is what's used by the `drake_ros::viz::RvizVisualizer` to transmit SceneGraph's shapes to Rviz.

As of https://github.com/RobotLocomotion/drake/issues/20932, the `drake::geometry::Mesh` shape in Drake is growing to allow for `*.gltf` files with external assets to be used (instead of only embedded assets).  In a glTF file with external assets, there are fields named `uri` where the value is a relative pathname.  We want those meshes to show up in Rviz properly.

In the `drake-ros` code, we convert a Drake `Mesh` shape to a `visualization_msgs::msg::Marker::MESH_RESOURCE`, like so:

```c++
  void ImplementGeometry(const drake::geometry::Mesh& mesh, ...) override {
    ...
    visualization_msgs::msg::Marker& marker = ...;
    marker.type = visualization_msgs::msg::Marker::MESH_RESOURCE;
    marker.mesh_resource = "file://" + mesh.filename();
```

That's enough to successfully send the glTF file to Rviz, and Rviz will try to load the file, but I'm pretty sure that Rviz won't show it correctly.  In similar experiments with other marker messages, when Rviz tries to load the cited `"uri": "foo.bin"` file containing the mesh geometry, it experiences some error.

The call to action here is to make `drake_ros::viz::RvizVisualizer` inter-operate with ROS 2 `rviz` so that at least the mesh geometry shows up.  (Ideally, the textures would show up as well, but the critical point is just the mesh geometry.)

That could happen either by fixing Rviz to load the glTF file properly even with external URIs, or by having `SceneMarkersSystem` rewrite the glTF file to embed the assets as `data:` URIs (using a tempfile).

jwnimmer-tri, 2024-02-21:
In terms of sample data for testing, see [drake/geometry/render/test/meshes](https://github.com/RobotLocomotion/drake/tree/master/geometry/render/test/meshes).  The files `cube{1,2,3}.gltf` have different asset-bundling techniques.  Ideally, they should all behave the same and show up as a cube shape in Rviz.

jwnimmer-tri, 2024-02-21:
I realized I can explain one thought slightly more carefully:

> ... or by having SceneMarkersSystem rewrite the glTF file to embed the assets ...

Specifically, if for example SceneMarkersSystem is given the `cube2.gltf` file it could internally reprocess it to look exactly like `cube1.gltf` instead, save that new file into `/tmp` somewhere, and then send that `file:///tmp/...` to Rviz.  Note that the only difference between cube1 to cube2 is the `uri` -- in one case it's a pathname to the asset, and in the other case it's a base64-encoded copy of the asset.

This approach of tempfile copying is of course somewhat gross, and it would better if Rviz did the right thing on the receiving end.  But that proves overly difficult, this work-around is pretty simple and should be very robust.  It's the approach Drake currently uses for one of our other shape consumers, the `RenderEngineGltfClient`:

https://github.com/RobotLocomotion/drake/blob/9450b456edd2727f7dee2518e8496a7e5c39c6ca/geometry/render_gltf_client/internal_merge_gltf.cc#L213

ahcorde, 2024-02-22:
Hi @jwnimmer-tri 

I was testing these models in RViz2. It looks that Assimp might have some bugs.

This is the current state of the cube models:
 - cube1: visualize mesh but for now we can't visualize texture

[image]

 - cube2: nothing
```bash
FileNotFoundException: Cannot locate resource file:///home/ahcorde/drake_stuff/drake/geometry/render/test/meshes/ in resource group rviz_rendering. in ResourceGroupManager::openResource at /home/ahcorde/ros2_humble/build/rviz_ogre_vendor/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceGroupManager.cpp (line 703)
```
The error is weird, `/home/ahcorde/drake_stuff/drake/geometry/render/test/meshes/` is a directory not a file.

 - cube3: nothing
```bash
GLTF: error while reading referenced file "cube3.bin"
```

ahcorde, 2024-02-22:
Related with the failured while reading a referenced file. I made this fix in assimp https://github.com/assimp/assimp/pull/5472

This allow the models to show the mesh at least in RViz, I'm digging in the texture issue

ahcorde, 2024-02-27:
I created this PR https://github.com/ros2/rviz/pull/1154 backporting the PR that you mentioned. This allows to at least load the meshes.

jwnimmer-tri, 2024-02-27:
Thanks!   And sorry I didn't mention that PR earlier in this thread.  It did not cross my mind that the RViz I was running might not have had that fix incorporated.

jwnimmer-tri, 2024-03-04:
I'd like to see https://github.com/ros2/rviz/pull/1154 be published into new Humble packages.

Do I need to do anything specific to request that to happen?

jwnimmer-tri, 2024-03-25:
Hmm, I didn't see any pull request at https://github.com/ros/rosdistro/pulls?q=is%3Apr+rviz2+is%3Aclosed related to this.  Did the new release happen yet?  Maybe I am looking in the wrong place?

jwnimmer-tri, 2024-04-12:
See that https://github.com/ros/rosdistro/pull/40329 merged a couple weeks ago, but I don't see the new package available yet when I upgrade.  How munch longer will this fix take to reach Humble users?

IanTheEngineer, 2024-04-15:
It looks like the `rosdistro` merge took place just a few days after the [latest Humble sync on 3-25](https://discourse.ros.org/t/preparing-for-humble-sync-2024-03-25/36776). These syncs tend to be every three weeks, so I expect there should be another one in the next week or so.

With that said, the folks running the ROS buildfarm are likely focused on the [Jazzy Jalisco release](https://docs.ros.org/en/rolling/Releases/Release-Jazzy-Jalisco.html#release-timeline) at the moment, with today being the API and ROS base freeze date. I am not sure if that has any impact on the Humble sync timelines, however.

jwnimmer-tri, 2024-04-29:
I think this is sufficiently finished.

closed 2024-04-29 by jwnimmer-tri