RobotLocomotion/drake-ros #347: problem while intergrate drake-ros with manipulation library on MIT manipulation course

levi3001, 2024-04-28:
I want to convert the code in MIT manipulation course to be run by ros2 using drake-ros repo. However, when I try to run the teleop example ([](https://github.com/RussTedrake/manipulation/blob/master/book/intro/intro.ipynb)), I encounter this error:

```
Traceback (most recent call last):
  File "/home/hoang/ws/install/drake_ros_examples/lib/drake_ros_examples/teleop.py", line 172, in <module>
    main()
  File "/home/hoang/ws/install/drake_ros_examples/lib/drake_ros_examples/teleop.py", line 120, in main
    station = builder.AddSystem(MakeHardwareStation(scenario, meshcat))
  File "/home/hoang/anaconda3/envs/robotics/lib/python3.10/site-packages/manipulation/station.py", line 932, in MakeHardwareStation
    added_models = ProcessModelDirectives(
RuntimeError: /home/hoang/anaconda3/envs/robotics/lib/python3.10/site-packages/manipulation/models/planar_iiwa14_no_collision.urdf:35: error: URI 'package://drake_models/iiwa_description/meshes/iiwa14/visual/link_0.gltf' resolved to '/home/hoang/.cache/drake/package_map/0a19e498d654e39dcf7e8f9144d06b3bbdd01a639cdc653a7d41cdc6d9ad5319-0ab6353f1af138d3c65dc47cdc797650d347b079e224bd519253a4d1b081c309/iiwa_description/meshes/iiwa14/visual/link_0.gltf' which does not exist.
[ros2run]: Process exited with failure 1
```
After some test, I realize that I only encounter this error after I source workspace to set up ros2 using this command:
`source ./ws/install/setup.bash`
If I do not source the set up file, the code works normally without any error.
Could you please explain while this issue happend and how to handle it?
Thank you

jwnimmer-tri, 2024-04-29:
That error message looks to me like the version of Drake being used is too old to contain that model asset.  That model was only updated to glTF form in the most recent release, Drake v1.28.0.

levi3001, 2024-04-30:
Thank you so much for your support, after I update drake, the model is finally loaded. 
It seems like when I installed the manipulation library, I installed the new realsease pydrake but when I installed ros-drake, I only installed the old version, that why I get confilct after soruce .ws/install/setup.bash