RobotLocomotion/drake-ros #358: Example simulation issue

seanres, 2024-07-05:
When attempting to run any of the example programs the URDF joint orientations are messed up. Currently using ROS2 Humble and Ubuntu 22.04. Any suggestions on what could be causing this or how to troubleshoot this issue?
[image]

miz2miz, 2024-09-30:
The same problem.

[image]

LiHaochuan123, 2025-03-24:
I met the same problem.

KhachDavid, 2025-04-23:
Was this issue ever resolved? Running on Humble and Ubuntu 22.04

[image]