RobotLocomotion/drake-ros #384: rmw_isolation_example_cc_test timeouts when running in CI

frneer, 2025-04-21:
This test timeouts when running in CI. Based on some testing the communication between talker and listener isn't working.

```
TIMEOUT: //ros2_example_apps:rmw_isolation_example_cc_test (Summary)
      /github/home/.cache/bazel/_bazel_root/9ffa39c466eb506336c2960019ab2bc0/execroot/ros2_example_bazel_installed/bazel-out/k8-fastbuild/testlogs/ros2_example_apps/rmw_isolation_example_cc_test/test.log
INFO: From Testing //ros2_example_apps:rmw_isolation_example_cc_test:
==================== Test output for //ros2_example_apps:rmw_isolation_example_cc_test:
Launching node talker
Launching node listener
[WARN] [1745260449.407230553] [rcl]: ROS_LOCALHOST_ONLY is deprecated but still honored if it is enabled. Use ROS_AUTOMATIC_DISCOVERY_RANGE and ROS_STATIC_PEERS instead.
[WARN] [1745260449.407265628] [rcl]: 'localhost_only' is enabled, 'automatic_discovery_range' and 'static_peers' will be ignored.
Waiting for listener to exit
[WARN] [1745260449.407960675] [rcl]: ROS_LOCALHOST_ONLY is deprecated but still honored if it is enabled. Use ROS_AUTOMATIC_DISCOVERY_RANGE and ROS_STATIC_PEERS instead.
[WARN] [1745260449.407983878] [rcl]: 'localhost_only' is enabled, 'automatic_discovery_range' and 'static_peers' will be ignored.
-- Test timed out at 2025-04-21 18:[35](https://github.com/RobotLocomotion/drake-ros/actions/runs/14578658364/job/40890313186#step:16:36):09 UTC --
[INFO] [1745260509.117506960] [rclcpp]: signal_handler(signum=15)
[INFO] [1745260509.117506054] [rclcpp]: signal_handler(signum=15)
======================================================================
```

Locally the test works and it's python version still works in CI (`rmw_isolation_example_py_test`).

frneer, 2025-04-22:
Same thing happens with `roslaunch_eg_test`

```
TIMEOUT: //ros2_example_apps:roslaunch_eg_test (Summary)
      /github/home/.cache/bazel/_bazel_root/9ffa39c466eb506336c2960019ab2bc0/execroot/ros2_example_bazel_installed/bazel-out/k8-fastbuild/testlogs/ros2_example_apps/roslaunch_eg_test/test.log
INFO: From Testing //ros2_example_apps:roslaunch_eg_test:
==================== Test output for //ros2_example_apps:roslaunch_eg_test:
[INFO] [launch]: All log files can be found below /github/home/.cache/bazel/_bazel_root/9ffa39c466eb506336c2960019ab2bc0/sandbox/processwrapper-sandbox/171/execroot/ros2_example_bazel_installed/_tmp/f54c77530ca0d8bca89e12e202068afa/fake_ros_home/log/2025-04-22-18-07-54-827747-e6b19cbaf848-36973
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [eg_talker-1]: process started with pid [37052]
[INFO] [eg_listener-2]: process started with pid [37053]
[eg_listener-2] [WARN] [1745345274.969635208] [rcl]: ROS_LOCALHOST_ONLY is deprecated but still honored if it is enabled. Use ROS_AUTOMATIC_DISCOVERY_RANGE and ROS_STATIC_PEERS instead.
[eg_listener-2] [WARN] [1745345274.969667468] [rcl]: 'localhost_only' is enabled, 'automatic_discovery_range' and 'static_peers' will be ignored.
[eg_talker-1] [WARN] [1745345275.874366727] [rcl]: ROS_LOCALHOST_ONLY is deprecated but still honored if it is enabled. Use ROS_AUTOMATIC_DISCOVERY_RANGE and ROS_STATIC_PEERS instead.
[eg_talker-1] [WARN] [1745345275.875180925] [rcl]: 'localhost_only' is enabled, 'automatic_discovery_range' and 'static_peers' will be ignored.
[eg_talker-1] [INFO] [1745345276.319[95](https://github.com/RobotLocomotion/drake-ros/actions/runs/14601485426/job/40960461617#step:16:96)1821] [node]: Publishing: 'Hello World: 0'
[eg_talker-1] [INFO] [1745345276.324646542] [node]: Publishing: 'Hello World: 1'
[eg_talker-1] [INFO] [1745345276.423969654] [node]: Publishing: 'Hello World: 2'
[eg_talker-1] [INFO] [1745345276.524174771] [node]: Publishing: 'Hello World: 3'
[eg_talker-1] [INFO] [1745345276.624323553] [node]: Publishing: 'Hello World: 4'
[eg_talker-1] [INFO] [1745345276.724322[98](https://github.com/RobotLocomotion/drake-ros/actions/runs/14601485426/job/40960461617#step:16:99)4] [node]: Publishing: 'Hello World: 5'
[eg_talker-1] [INFO] [1745345276.824603375] [node]: Publishing: 'Hello World: 6'
[eg_talker-1] [INFO] [1745345276.924399355] [node]: Publishing: 'Hello World: 7'
[eg_talker-1] [INFO] [1745345277.024368759] [node]: Publishing: 'Hello World: 8'
[eg_talker-1] [INFO] [1745345277.124871152] [node]: Publishing: 'Hello World: 9'
[INFO] [eg_talker-1]: process has finished cleanly [pid 37052]
-- Test timed out at 2025-04-22 18:12:53 UTC --
================================================================================
```

ggould-tri, 2025-04-24:
For attention of @calderpg-tri (this was flagged for you in the review but I'm about to close out that review)