RobotLocomotion/drake-ros #385: drake builds fail if gcc14 is installed in the system

frneer, 2025-04-22:
Drake's mkdoc script loads gcc14 libraries if installed which causes the build to fail because it expects gcc13 to be used.

[This target](https://github.com/RobotLocomotion/drake/blob/17f37909acc2c75d19251568af986df9f37a12d4/bindings/pydrake/BUILD.bazel#L56-L65) fails when building it with the following error:

```
/../lib/gcc/x86_64-linux-gnu/14/../../../../include/c++/14/bits/iterator_concepts.h:995:16: note: because 'is_array_v<std::__unicode::_Utf_view<char, const char32_t (&)[1]> >' evaluated to false
/../lib/gcc/x86_64-linux-gnu/14/../../../../include/c++/14/bits/iterator_concepts.h:995:35: note: and 'std::__unicode::_Utf_view<char, const char32_t (&)[1]> &' does not satisfy '__member_begin'
/../lib/gcc/x86_64-linux-gnu/14/../../../../include/c++/14/bits/iterator_concepts.h:979:23: note: because '__decay_copy(__t.begin())' would be invalid: no member named 'begin' in 'std::__unicode::_Utf_view<char, const char32_t (&)[1]>'
/../lib/gcc/x86_64-linux-gnu/14/../../../../include/c++/14/bits/iterator_concepts.h:995:59: note: and 'std::__unicode::_Utf_view<char, const char32_t (&)[1]> &' does not satisfy '__adl_begin'
/../lib/gcc/x86_64-linux-gnu/14/../../../../include/c++/14/bits/iterator_concepts.h:989:19: note: because '__decay_copy(begin(__t))' would be invalid: no matching function for call to 'begin'
fatal error: too many errors emitted, stopping now [-ferror-limit=]
Traceback (most recent call last):
  File "/home/runner/.cache/bazel/_bazel_runner/d07b19e4bc14c7eead45599d9bb941ba/sandbox/processwrapper-sandbox/2387/execroot/drake_ros/bazel-out/k8-opt-exec-ST-a828a81199fe/bin/external/drake/tools/workspace/pybind11/mkdoc.runfiles/drake_ros/../drake/tools/workspace/pybind11/mkdoc.py", line 815, in <module>
    main()
  File "/home/runner/.cache/bazel/_bazel_runner/d07b19e4bc14c7eead45599d9bb941ba/sandbox/processwrapper-sandbox/2387/execroot/drake_ros/bazel-out/k8-opt-exec-ST-a828a81199fe/bin/external/drake/tools/workspace/pybind11/mkdoc.runfiles/drake_ros/../drake/tools/workspace/pybind11/mkdoc.py", line 781, in main
    raise RuntimeError(
RuntimeError: Parsing headers using the clang library failed with 19 error(s) and 1 fatal error(s)
```

If gcc14 is purged, the build is succesful.

jwnimmer-tri, 2025-04-22:
FYI see https://github.com/RobotLocomotion/drake/issues/22905.

jwnimmer-tri, 2025-05-16:
Once the #389 upgrade happens, the fix for https://github.com/RobotLocomotion/drake/issues/22905 will included, so hopefully this problem will be resolved.  (Assuming that drake-ros calls drake's install_prereqs before building Drake.)

closed 2025-07-30 by ggould-tri